zeule/asus-ec-sensors #3: VRM temperature sensor reading are incorrect

zeule, 2021-11-27:
Right after system start, the supposed to be VRM temperature sensor (0x003e) readings are lower than the ambient temperature.

zeule, 2022-02-14:
DSDT codes for AMD 400-series motherboards seem to read the CPU VRM temperature value in exactly the same way as we do: single-byte value at 0x3e in degree Celsius.

pheiduck, 2022-06-12:
At my PRIME X570-Pro 
VRM is at 0°C? Is this correct or do this board do not have a sensor for that?

zeule, 2022-06-12:
Usual blank value for temperature sensors is -40 ℃, but of course constant value of 0 suggests there is no real data there. Unfortunately, from a glance I can't find whom sensors definitions for this board came from, and can't ask that person for feedback.

pheiduck, 2022-06-19:
But the thing is that the VRM temps are displayed on hwinfo (Windows). So I think they are not read here?

zeule, 2022-06-19:
HWINFO groups sensors by source. If the VRM temp is listed inside the ASUS EC group, HWINFO probably reads it from EC. Perhaps for this model the register address for VRM temp differs from other X570 boards? I still don't know how does ASUS organise things.

RobotRoss, 2024-07-09:
I've also got issues with this. The VRM temp sensor is stuck at +20.0 degrees C regardless of the actual VRM temp. X570-E GAMING here.

zeule, 2024-07-11:
> I've also got issues with this. The VRM temp sensor is stuck at +20.0 degrees C regardless of the actual VRM temp. X570-E GAMING here.

Does ASUS software show another value for VRM temperature? Or any other software?

RobotRoss, 2024-07-11:
> > I've also got issues with this. The VRM temp sensor is stuck at +20.0 degrees C regardless of the actual VRM temp. X570-E GAMING here.
> 
> Does ASUS software show another value for VRM temperature? Or any other software?

Apologies, seems like my board doesn't actually have a VRM temp sensor, so it should probably be hidden instead.

> ASUS has neglected to utilize an integrated thermal sensor on all but its top tier AM4 models which is disappointing.
https://www.anandtech.com/show/16348/asus-rog-strix-x570e-gaming-motherboard-review/9

zeule, 2024-07-11:
Thanks! Would you like to make the change yourselves?

RobotRoss, 2024-07-11:
> Thanks! Would you like to make the change yourselves?

I think this is correct? (First time reading the source)
https://github.com/zeule/asus-ec-sensors/pull/58

zeule, 2024-07-11:
Yes, thank you!