zeule/asus-ec-sensors #10: Request: Add X570-F board

olehj, 2022-01-17:
Board: "ROG STRIX X570-F GAMING" - uses the same \\AMW0.ASMX as the other boards

Sensors verified to work:
SENSOR_SET_TEMP_CHIPSET_CPU_MB
SENSOR_TEMP_T_SENSOR
SENSOR_FAN_CHIPSET

asus-ec-sensors-isa-0000
Adapter: ISA adapter
Chipset:     2672 RPM
Chipset:      +61.0°C  
CPU:          +34.0°C  
Motherboard:  +35.0°C  
T_Sensor:     +30.0°C  
VRM:           +0.0°C  
CPU:           0.00 A

zeule, 2022-01-17:
Thank you! The VRM temperature sensors shows 0 in your example. Shouldn't it be removed? Maybe you can test with ASUS software (or another software like HWINFO) on Windows?

olehj, 2022-01-17:
I posted the results with a direct copy from the E-board just modifying it to F. There's no VRM temp nor CPU current as far as I could see in HWINFO or even the UEFI directly.

So the three sensor variables in the initial posts seems to be the working ones (I used the review2 branch).

zeule, 2022-01-17:
OK, so we remove the VRM temp. But I'm surprised there is no CPU current sensor. Could you, please, test under a high CPU load?

olehj, 2022-01-17:
No info (still 0A) using Prime95 full load on all 12/24 cores/ht. AMD 5900X

olehj, 2022-01-17:
...dunno if it requires some C-states or other ACPI features from UEFI which I might have disabled?

zeule, 2022-01-17:
Thank you! I will add the board with the provided information.

commit d45be0a2 (2022-01-18): Add ROG STRIX X570-F GAMING

Closes #10.
closed 2022-01-18 by zeule