zeule/asus-ec-sensors #14: asus-ec-sensors won't compile

KeithMyers, 2022-02-08:
Do you know why your module won't compile?
All I get and another friend gets is this error message.

```
keith@Serenity:~/Downloads/asus-ec-sensors-master$ make
  CC [M]  /home/keith/Downloads/asus-ec-sensors-master/asus-ec-sensors.o
/home/keith/Downloads/asus-ec-sensors-master/asus-ec-sensors.c: In function ‘scale_sensor_value’:
/home/keith/Downloads/asus-ec-sensors-master/asus-ec-sensors.c:481:18: error: ‘MILLI’ undeclared (first use in this function)
  481 |   return value * MILLI;
      |                  ^~~~~
/home/keith/Downloads/asus-ec-sensors-master/asus-ec-sensors.c:481:18: note: each undeclared identifier is reported only once for each function it appears in
make[2]: *** [scripts/Makefile.build:281: /home/keith/Downloads/asus-ec-sensors-master/asus-ec-sensors.o] Error 1
make[1]: *** [Makefile:1879: /home/keith/Downloads/asus-ec-sensors-master] Error 2
make: *** [Makefile:32: modules] Error 2
```
keith@Serenity:~/Downloads/asus-ec-sensors-master$

zeule, 2022-02-08:
`MILLI` appeared in kernel 5.15, you probably use an older one. I've added a workaround.

KeithMyers, 2022-02-08:
Thank you for the quick fix.  It now compiles but is is producing a warning now.  Safe to ignore or not?
keith@Serenity:~/Downloads/asus-ec-sensors-master$ 
```
make
  CC [M]  /home/keith/Downloads/asus-ec-sensors-master/asus-ec-sensors.o
  MODPOST /home/keith/Downloads/asus-ec-sensors-master/Module.symvers
  CC [M]  /home/keith/Downloads/asus-ec-sensors-master/asus-ec-sensors.mod.o
  LD [M]  /home/keith/Downloads/asus-ec-sensors-master/asus-ec-sensors.ko
  BTF [M] /home/keith/Downloads/asus-ec-sensors-master/asus-ec-sensors.ko
Skipping BTF generation for /home/keith/Downloads/asus-ec-sensors-master/asus-ec-sensors.ko due to unavailability of vmlinux

keith@Serenity:~/Downloads/asus-ec-sensors-master$
 ls
asus-ec-sensors.c    asus-ec-sensors.mod.c  dkms.conf.am  Makefile        README.md
asus-ec-sensors.ko   asus-ec-sensors.mod.o  Kbuild        modules.order
asus-ec-sensors.mod  asus-ec-sensors.o      LICENSE       Module.symvers

```
keith@Serenity:~/Downloads/asus-ec-sensors-master$

zeule, 2022-02-08:
I don't really have any experience with distributions where development files are in detached packages. The standard minimal makefile here assumes it can find kernel build tree at /lib/modules/$(uname -r)/build. Maybe installing dkms can help, because it has to pull in everything required to compile out-of-tree modules?

KeithMyers, 2022-02-08:
DKMS is installed.

zeule, 2022-02-08:
OK, then what do you find at /lib/modules/$(uname -r)/build? What distribution are you using?

KeithMyers, 2022-02-08:
```
keith@Serenity:/lib/modules/5.13.0-28-generic/build$ ls
arch    Documentation  init     kernel    Module.symvers  security  usr
block   drivers        ipc      lib       net             sound     virt
certs   fs             Kbuild   Makefile  samples         tools
crypto  include        Kconfig  mm        scripts         ubuntu

```
I am running Ubuntu 20.04.3 LTS HWE kernel 5.13.0-28

KeithMyers, 2022-02-08:
Just tried to install the module.
```
sudo make dkms
[sudo] password for keith: 

Creating symlink /var/lib/dkms/asus-ec-sensors/unknown/source ->
                 /usr/src/asus-ec-sensors-unknown

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j32 KERNELRELEASE=5.13.0-28-generic KVER=5.13.0-28-generic DKMS_BUILD=1...
Signing module:
 - /var/lib/dkms/asus-ec-sensors/unknown/5.13.0-28-generic/x86_64/module/asus-ec-sensors.ko
Secure Boot not enabled on this system.
cleaning build area...

DKMS: build completed.

asus-ec-sensors.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.13.0-28-generic/updates/dkms/

depmod...

DKMS: install completed.
modprobe: ERROR: could not insert 'asus_ec_sensors': No such device
make: *** [Makefile:47: dkms] Error 1
```

zeule, 2022-02-08:
So, DKMS knowns where to find vmlinux. 

> modprobe: ERROR: could not insert 'asus_ec_sensors': No such device

This means the driver does not recognise your mainboard model.

KeithMyers, 2022-02-08:
OK.  Thanks.

zeule, 2022-02-08:
If you have reasons to believe the board has this EC interface, we can add support for this model with your help.

KeithMyers, 2022-02-08:
My board does not match the makefile. C7H.  I already use the asus-wmi-sensors driver.
 I am trying to help a friend with a ASUS Crosshair VIII Hero board install the driver.  He is running the same distro and kernel.
It should install on his board fine.

zeule, 2022-02-08:
Well, in principle the should work with C7H, and in #12 we see it works with X470. And there is a questions to sensor addresses for X470, which differs from the 500 series, as @darcagn found for a particular mode. But we don't know yet is it the case for other 400-series boards. Thus I would appreciate if you find a time to make a change to the driver to load it on your system and test if the value reported as Water_Flow is in fact CPU_Opt on C7H.

KeithMyers, 2022-02-08:
Sure I'll give it a try.  But as you posted it will not recognize my board during the modprobe installation.
I see in [#12](https://github.com/zeule/asus-ec-sensors/issues/12)
I will have to add my Crosshair VII Hero board to the asus_ec_dmi_table.  Can you tell me where that resides?
Haven't found it yet.

KeithMyers, 2022-02-08:
I got the dsdt.dsl output.  On my C7H board the Water Flow sensor is at:
 Package (0x06)
                {
                    "Water Flow", 
                    0x04, 
                    0x07, 
                    0x02, 
                    One, 
                    0x1B
                }, 

Sensors -u shows the native enumeration as Fan 8

darcagn, 2022-02-08:
> Sure I'll give it a try. But as you posted it will not recognize my board during the modprobe installation. I see in [#12](https://github.com/zeule/asus-ec-sensors/issues/12) I will have to add my [...]

Go to the line `static const struct dmi_system_id asus_ec_dmi_table[] __initconst = {` (current revision it's at line 179)
Copy one of the entries that already exist, but you can add/remove sensors to try. 
Change the name of the board (in quotes) with the name of your board (use `cat /sys/class/dmi/id/board_name` to get that exact name).

KeithMyers, 2022-02-08:
Added my board to the asus-ec-sensors.c file and compiled and installed the new driver. The sensors output is very limited compared to the asus-wmi-sensors output with many erroneous values.
```
 sensors
asusec-isa-0000
Adapter: ISA adapter
CPU Core:      1.25 V  
CPU_Opt:        0 RPM
Chipset:        0 RPM
Water_Flow:     0 RPM
Chipset:      +51.0°C  
CPU:          +83.0°C  
Motherboard:  +36.0°C  
T_Sensor:     -40.0°C  
VRM:          +65.0°C  
Water_In:      +0.0°C  
Water_Out:    -40.0°C  
CPU:         -119.00 A  
```
Also here is the output of asus-wmi-sensors.

```
asuswmisensors-isa-0000
Adapter: ISA adapter
CPU Core Voltage:          1.23 V  
CPU SOC Voltage:           1.08 V  
DRAM Voltage:              1.41 V  
VDDP Voltage:            665.00 mV 
1.8V PLL Voltage:          1.81 V  
+12V Voltage:             11.88 V  
+5V Voltage:               4.99 V  
3VSB Voltage:              3.33 V  
VBAT Voltage:              3.18 V  
AVCC3 Voltage:             3.31 V  
SB 1.05V Voltage:          1.08 V  
CPU Core Voltage:          1.25 V  
CPU SOC Voltage:           1.09 V  
DRAM Voltage:              1.46 V  
CPU Fan:                 1901 RPM
Chassis Fan 1:              0 RPM
Chassis Fan 2:              0 RPM
Chassis Fan 3:              0 RPM
HAMP Fan:                   0 RPM
Water Pump:                 0 RPM
CPU OPT:                    0 RPM
Water Flow:                 0 RPM
AIO Pump:                   0 RPM
CPU Temperature:          +83.0°C  
CPU Socket Temperature:   +59.0°C  
Motherboard Temperature:  +36.0°C  
Chipset Temperature:      +51.0°C  
Tsensor 1 Temperature:   +216.0°C  
CPU VRM Temperature:      +65.0°C  
Water In:                +216.0°C  
Water Out:                +37.0°C  
CPU VRM Output Current:  137.00 A 
```
Cpu Fan is missing and Water Out is not scaled correctly. Neither is CPU VRM Output Current.

KeithMyers, 2022-02-08:
I copied the ROG Crosshair VIII Hero block for clarification and just changed the name. Figured it was the closest to the C7H.

zeule, 2022-02-08:
Thank you, this is very helpful! 

> The sensors output is very limited compared to the asus-wmi-sensors output with many erroneous values.

asus-wmi-sensors takes information from the board WMI code, which collects data from the Super I/O chip and the EC registers and provides the full set. This driver only queries the EC. Unfortunately, the ASUS WMI code is buggy for some boards and this driver can provide an option for those affected.

I see that you do not have CPU_Opt fan connected, but you happen to use the Water out sensor. We see that addresses for water do not match (Water Out at EC shows blank, which is either Water In or T_Sensor, but likely Water In as T_Sensor is also blank in both cases), I see that CPU current at EC is CPU VRM Output Current but I read it as signed while it is unsigned in fact. All that is valuable information for this driver, thank you very much for providing it!

If you want to further contribute, maybe we can try to find the correct Water Out sensor address? I suppose it is not far from what is called now Water In, which seems to be Water Out for you. 

@darcagn, I conclude that for the 400-series I need an independent set of sensor definitions and also have to correct signess for the CPU current (have to encode it in the sensor address). Thus, adding support for your board will take some days.

KeithMyers, 2022-02-08:
Is this what you are looking for?
 Package (0x06)
                {
                    "Water Out", 
                    One, 
                    0x07, 
                    0x02, 
                    Zero, 
                    0x1A
                },

zeule, 2022-02-08:
Maybe yes, but I don't know what these number mean. Could you share the whole dsdt file (binary or decompiled)?

KeithMyers, 2022-02-08:
I only use the Water Out sensor as it is easiest to get to and not buried under a graphics card being on the front of the board.
I only have the CPU Fans connected since I am all water cooled and moved to a fan hub back in the day when the C7H BIOS was having issues with the WMI output and the fan header outputs going missing entirely with overheated computers. 

They fixed that issue in later BIOS', I don't remember at what point in the BIOS series evolution but by that time I had already moved to a fan hub and never felt the need to move back to on board fan headers.

I do still have one C7H host that is running the fans off the board fan headers. That host would be a better candidate for testing the asus-ec-sensors driver on I suppose.

I run all my fans at 100% maximum output all the time since all hosts are always 100% engaged in BOINC computing 24/7.
Nothing like having to turn on the air conditioning today in the middle of winter when it is 80° F. outside.

KeithMyers, 2022-02-09:
Sure here is the binary dsdt.dat file
[dsdt.zip](https://github.com/zeule/asus-ec-sensors/files/8028339/dsdt.zip)

zeule, 2022-02-09:
@darcagn 
Found them all. In blocks like this 
```
Package (0x06)
{
    "CPU VRM Output Current", 
    0x03, 
    0x06, 
    0x02, 
    One, 
    0x17
}, 
```
the last number (here 0x17) is the index in the `SENB` array, and its records (NUxx) are assigned inside the `UPEC` method. Will trace them all tomorrow.

KeithMyers, 2022-02-09:
I have finally got my friend to successfully install the asus-ec-sensors driver on his ROG CROSSHAIR VIII HERO motherboard.

But I am really disappointed in the number of sensors exposed.  I thought this would have been sorted out a long while ago. I assumed that you had developed on the C8H board or had somebody help out by now.

Only two fans exposed compared to the 3 fans connected as shown by the SIO chip NCT6798 driver.
At least the temps look correct.  Only one voltage? Remembering what the sensor names were added to the source file for my C7H attempt, I guess that is all that can be expected.

Would a dsdt.dat dump file help get more sensors exposed? I can have my friend get that for you,

KeithMyers, 2022-02-09:
This is the output on the C8H board.
george@GWG-PC3:~/Downloads/asus-ec-sensors-master$ 
```
sensors
asusec-isa-0000
Adapter: ISA adapter
CPU Core:      1.09 V  
CPU_Opt:     1733 RPM
Chipset:     2977 RPM
Water_Flow:     0 RPM
Chipset:      +67.0°C  
CPU:          +54.0°C  
Motherboard:  +36.0°C  
T_Sensor:     +36.0°C  
VRM:          +42.0°C  
Water_In:     -40.0°C  
Water_Out:    -40.0°C  
CPU:         102.00 A  
```
And this is the SIO chip output from the NCT6798 driver.

```
nct6798-isa-0290
Adapter: ISA adapter
in0:                        1.09 V  (min =  +0.00 V, max =  +1.74 V)
in1:                      1000.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in2:                        3.39 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in3:                        3.28 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in4:                        1.73 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in5:                      592.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in6:                        1.05 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in7:                        3.38 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in8:                        3.33 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in9:                        1.03 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in10:                      88.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in11:                     104.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in12:                       1.02 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in13:                       1.49 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in14:                       1.03 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
fan1:                     1236 RPM  (min =    0 RPM)
fan2:                     1689 RPM  (min =    0 RPM)
fan3:                        0 RPM  (min =    0 RPM)
fan4:                     1496 RPM  (min =    0 RPM)
fan5:                        0 RPM  (min =    0 RPM)
fan6:                        0 RPM  (min =    0 RPM)
fan7:                        0 RPM  (min =    0 RPM)
SYSTIN:                    +36.0°C  (high = +80.0°C, hyst = +75.0°C)  sensor = thermistor
CPUTIN:                    +52.5°C  (high = +80.0°C, hyst = +75.0°C)  sensor = thermistor
AUXTIN0:                   +23.0°C    sensor = thermistor
AUXTIN1:                  +101.0°C    sensor = thermistor
AUXTIN2:                   +98.0°C    sensor = thermistor
AUXTIN3:                   +24.0°C    sensor = thermistor
PECI Agent 0 Calibration:  +54.5°C  
PCH_CHIP_CPU_MAX_TEMP:      +0.0°C  
PCH_CHIP_TEMP:              +0.0°C  
PCH_CPU_TEMP:               +0.0°C  
intrusion0:               ALARM
intrusion1:               ALARM
beep_enable:              disabled
```

```

KeithMyers, 2022-02-09:
These are his fans running at 100% of their designed speeds.
fan1:                     1236 RPM  (min =    0 RPM)   The rear exhaust fan (typical case fan)
fan2:                     1689 RPM  (min =    0 RPM)   The top rad fans (Arctic P12 fans on the rad)
fan3:                        0 RPM  (min =    0 RPM)
fan4:                     1496 RPM  (min =    0 RPM)   The front Noctua fans (NF-A14)

George396, 2022-02-09:
From the Crosshair VIII Hero board.
[dsdt.dat.gz](https://github.com/zeule/asus-ec-sensors/files/8029087/dsdt.dat.gz)

KeithMyers, 2022-02-09:
Thanks for the upload of the dsdt file for your board George.
Should be helpful for the developer.

George396, 2022-02-09:
Thank you for all your hard work!  I couldn't have done it without your help.

zeule, 2022-02-09:
> But I am really disappointed in the number of sensors exposed. I thought this would have been sorted out a long while ago. I assumed that you had developed on the C8H board or had somebody help out by now.

My understanding is that the sensor set exposed via EC is complimentary to the one, provided by the Super I/O chip (nct6798d in case of C8H). My main motivation when I started this project was to find out how to read water and the T_Sensor temperature sensors, and every other found sensor is a by-product. Your assumption that it all began with C8H is correct.

KeithMyers, 2022-02-09:
So you have no intention of exposing any more of the sensors exposed by the nct6798d SIO chip?  Not even the case fans?

KeithMyers, 2022-02-09:
I just read this statement by you in the other issue.
**this driver does not replace the nctXXX one, but rather complements it with other sensor reading.** 
This statement does not appear in your readme on the front page. 
My impression was this was a _replacement_ for the SIO driver since that driver does not expose all the sensors and this new driver was the remedy.

zeule, 2022-02-10:
> This statement does not appear in your readme on the front page.

I believe this very first sentence in the readme is fully equivalent:
"Many ASUS motherboards do not publish all the available sensors via the Super I/O chip but the missing ones are available through the embedded controller (EC) registers."

> So you have no intention of exposing any more of the sensors exposed by the nct6798d SIO chip? Not even the case fans?

I exposed every sensor I'm aware of. If we discover more, I will be happy to add them. If some of them repeats the ones from the SIO chip, I might hide them under a module option.

KeithMyers, 2022-02-10:
OK, understood.  I just thought this driver was like a parallel development of the asus-wmi-sensors driver which completely replaced the buggy and deprecated it87 driver for the X470 boards. Plus the default drivers are never scaled correctly so mostly meaningless without a lot of extra work writing config files.

The full sensor set is exposed by the asus-wmi-sensors driver.  I thought that was your same goal.

zeule, 2022-02-10:
> The full sensor set is exposed by the asus-wmi-sensors driver.

@darcagn showed that this is not entirely true, when they found that the CPU current sensor for their board works, but is absent from the ASUS WMI hardware monitoring set, which is the base for asus_wmi_sensors. 

BTW, reading via the WMI functions was terribly slow in the previous version of this driver. I wonder how is that with the asus_wmi_sensors? How much time does it take to read all of them? `time sensors 'asuswmisensors-*'`?

KeithMyers, 2022-02-10:
Reads all the sensors on my boards.  And is fast too. And all the sensor values are 100% correct.
```
keith@Serenity:~$ time sensors 'asuswmisensors-*'
asuswmisensors-isa-0000
Adapter: ISA adapter
CPU Core Voltage:          1.23 V  
CPU SOC Voltage:           1.08 V  
DRAM Voltage:              1.41 V  
VDDP Voltage:            600.00 mV 
1.8V PLL Voltage:          1.83 V  
+12V Voltage:             11.88 V  
+5V Voltage:               4.99 V  
3VSB Voltage:              3.33 V  
VBAT Voltage:              3.18 V  
AVCC3 Voltage:             3.33 V  
SB 1.05V Voltage:          1.08 V  
CPU Core Voltage:          1.25 V  
CPU SOC Voltage:           1.09 V  
DRAM Voltage:              1.46 V  
CPU Fan:                 1880 RPM
Chassis Fan 1:              0 RPM
Chassis Fan 2:              0 RPM
Chassis Fan 3:              0 RPM
HAMP Fan:                   0 RPM
Water Pump:                 0 RPM
CPU OPT:                    0 RPM
Water Flow:                 0 RPM
AIO Pump:                   0 RPM
CPU Temperature:          +74.0°C  
CPU Socket Temperature:   +51.0°C  
Motherboard Temperature:  +30.0°C  
Chipset Temperature:      +45.0°C  
Tsensor 1 Temperature:   +216.0°C  
CPU VRM Temperature:      +57.0°C  
Water In:                +216.0°C  
Water Out:                +31.0°C  
CPU VRM Output Current:  128.00 A  

real	0m0.534s
user	0m0.000s
sys	0m0.016s
keith@Serenity:~$
```

bedaro, 2022-03-06:
The latest commit a74fa23153651ef0fa05d9d551764215f2d2bd57 seems to have broken the MILLI workaround on at least some systems (like mine, Ubuntu 20.04.4 LTS). The problem appears to be a bash-ism in the Makefile (subbing 5.13.0 as KVER_NUM):

`echo -e "5.15\n5.13.0" | sort -Ct. -k1,1n -k2,2n && echo YES`

When run in bash, this evaluates as empty as intended. Ubuntu uses dash as its implementation of sh, however, and its builtin echo does not recognize the -e option:

```
$ /bin/sh -c 'echo -e "5.15\n5.13.0"'
-e 5.15
5.13.0
```

I would suggest using `/bin/echo` instead of `echo` to avoid the unpredictable builtin behavior.

zeule, 2022-03-06:
Thank you, @bedaro!

closed 2022-04-27 by zeule