zeule/asus-ec-sensors #15: Crosshair VI Hero WiFi AC (X370) Support

darcagn, 2022-02-09:
board_name: `ROG CROSSHAIR VI HERO (WI-FI AC)`
mutex name: `\AMW0.ASMX` (same as others)

This board is being used in an Ubuntu LTS server with an older 5.4 kernel -- it was unable to compile at first because linux/dev_printk.h and linux/units.h weren't found. I removed the includes for those two header files and it seems to compile and work just fine, but I am unsure if functionality has been affected. 

These sensors work just fine and match the same output from asus-wmi-sensors:
`SENSOR_SET_TEMP_CHIPSET_CPU_MB | SENSOR_TEMP_T_SENSOR | SENSOR_CURR_CPU`

`SENSOR_FAN_CPU_OPT_X470` is the same situation as the X470 board, CPU_OPT is at 0x00bc (in place of Water_Flow for 500-series).

I plugged in fans/probes to the other sensor connectors and could not get a reading from any other sensor, including CPU voltage. I didn't get to poke around the registers with this board though, so there may be other undiscovered sensors exposed in the EC.

I have attached the DSDT file for this board if it helps. 
[ROG CROSSHAIR VI HERO (WI-FI AC).zip](https://github.com/zeule/asus-ec-sensors/files/8028399/ROG.CROSSHAIR.VI.HERO.WI-FI.AC.zip)

Example output with fan plugged into CPU_OPT and temp probe attached to T_SENSOR:
```
asusec-isa-0000
Adapter: ISA adapter
CPU Core:      0.00 V  
CPU_Opt:        0 RPM
VRM HS:         0 RPM
Chipset:        0 RPM
Water_Flow:  1336 RPM
Chipset:      +53.0°C  
CPU:          +45.0°C  
Motherboard:  +34.0°C  
T_Sensor:     +27.0°C  
VRM:          +35.0°C  
Water_In:      +0.0°C  
Water_Out:     +0.0°C  
CPU:           2.00 A  

asuswmisensors-isa-0000
Adapter: ISA adapter
CPU Core Voltage:          1.53 V  
CPU SOC Voltage:           1.13 V  
DRAM Voltage:              1.37 V  
VDDP Voltage:            916.00 mV 
1.8V PLL Voltage:          1.83 V  
+12V Voltage:             12.10 V  
+5V Voltage:               4.99 V  
3VSB Voltage:              3.31 V  
VBAT Voltage:              3.14 V  
AVCC3 Voltage:             3.33 V  
SB 1.05V Voltage:          1.05 V  
CPU Fan:                 1824 RPM
Chassis Fan 1:            876 RPM
Chassis Fan 2:            854 RPM
Chassis Fan 3:            854 RPM
AIO Pump:                   0 RPM
Water Pump:                 0 RPM
CPU OPT:                 1336 RPM
Water Flow:                 0 RPM
CPU Temperature:          +45.0°C  
CPU Socket Temperature:   +36.0°C  
Motherboard Temperature:  +34.0°C  
Chipset Temperature:      +53.0°C  
Tsensor 1 Temperature:    +27.0°C  
CPU VRM Temperature:      +35.0°C  
Water In:                +216.0°C  
Water Out:               +216.0°C  
CPU VRM Output Current:    2.00 A  
```

zeule, 2022-02-10:
Summary for this board.

The `UPEC` method fills in 9 sensors, numbered 19 to 27 (0x13 to 0x1b):
```
Method (UPEC, 0, Serialized)
{
	If ((Acquire (_GL, 0x03E8) == Zero))
	{
		WECB (Zero)
		Local0 = REC1 (0x90)
		Local1 = REC1 (0x91)
		Local0 = ((Local0 << 0x08) | Local1)
		Local0 *= 0x0C35
		NU19 = Local0
		Local0 = REC1 (0x3A)
		NU20 = Local0
		Local0 = REC1 (0x3D)
		NU21 = Local0
		Local0 = REC1 (0x3E)
		NU22 = Local0
		Local0 = REC1 (0xF4)
		NU23 = Local0
		Local0 = REC1 (0xBC)
		Local1 = REC1 (0xBD)
		Local0 = ((Local0 << 0x08) | Local1)
		NU24 = Local0
		Local0 = REC2 (One, 0x0D)
		NU25 = Local0
		Local0 = REC2 (One, 0x0B)
		NU26 = Local0
		Local0 = REC2 (Zero, 0xB4)
		Local1 = REC2 (Zero, 0xB5)
		Local0 = ((Local0 << 0x08) | Local1)
		NU27 = Local0
		Release (_GL)
		Return (Zero)
	}

	Return (Ones)
}
```
Corresponding records from the `INFO` package:
```
Package (0x06)
{
	"SB 1.05V Voltage", 
	Zero, 
	0x03, 
	0x02, 
	0x03, 
	0x13
}, 

Package (0x06)
{
	"Chipset Temperature", 
	One, 
	0x04, 
	0x02, 
	Zero, 
	0x14
}, 

Package (0x06)
{
	"Tsensor 1 Temperature", 
	One, 
	0x05, 
	0x02, 
	Zero, 
	0x15
}, 

Package (0x06)
{
	"CPU VRM Temperature", 
	One, 
	0x06, 
	0x02, 
	Zero, 
	0x16
}, 

Package (0x06)
{
	"CPU VRM Output Current", 
	0x03, 
	0x06, 
	0x02, 
	One, 
	0x17
}, 

Package (0x06)
{
	"CPU OPT", 
	0x02, 
	0x03, 
	0x02, 
	One, 
	0x18
}, 

Package (0x06)
{
	"Water In", 
	One, 
	0x07, 
	0x02, 
	Zero, 
	0x19
}, 

Package (0x06)
{
	"Water Out", 
	One, 
	0x07, 
	0x02, 
	Zero, 
	0x1A
}, 

Package (0x06)
{
	"Water Flow", 
	0x04, 
	0x07, 
	0x02, 
	One, 
	0x1B
}
```

|Label | source | size | address| notes|
|-------|-----------|------|------------| ------ |
| SB 1.05V Voltage | REC1 | 2 | 0x90 | *= 0x0C35 |
| Chipset Temperature | REC1 | 1 | 0x3a| |
| Tsensor 1 Temperature | REC1 | 1 | 0x3d | |
| CPU VRM Temperature | REC1 | 1 | 0x3e | |
| CPU VRM Output Current | REC1 | 1 | 0xf4| |
| CPU OPT | REC1 | 2 | 0xbc| |
| Water In | REC2 | 1| 0x01, 0x0d | |
| Water Out | REC2 | 1 | 0x01, 0x0b| |
| Water Flow | REC2 | 2| 0xb4| |

Now, that does `REC2()` do?

darcagn, 2022-02-10:
Recompiling and testing with these new values:

- SB 1.05V Voltage shows up, but it's only showing about 338mV when asus-wmi-sensors shows 1.06V. I don't know what the *=0x0C35 means, though.
- Chipset Temp, Tsensor 1 Temp, CPU VRM Temp, CPU VRM Current, and CPU OPT values are working properly and agree with asus-wmi-sensors.
- The three sensors showing REC2 don't work, I guess something else has to be changed.

zeule, 2022-02-10:
> Now, what does REC2() do?

Here is its code:
```
Method (REC2, 2, Serialized)
{
    IBFY ()
    EC6C = 0x91
    IBFY ()
    EC68 = 0xFF
    IBFY ()
    EC68 = Arg0
    IBFY ()
    EC6C = 0x90
    IBFY ()
    EC68 = Arg1
    OBFY ()
    Local0 = EC68 /* \AMW0.EC68 */
    Return (Local0)
}
```

Looks like it reads from a EC controller via ports 0x68 and 0x6c, but what are constants 0x91 and 0x90 for?

zeule, 2022-02-10:
> * SB 1.05V Voltage shows up, but it's only showing about 338mV when asus-wmi-sensors shows 1.06V. I don't know what the *=0x0C35 means, though.

I guess it is a conversion factor to volts. 0x0C35 (=3125) * 0.228 = 1.056, which is pretty close to 1.06.

> * Chipset Temp, Tsensor 1 Temp, CPU VRM Temp, CPU VRM Current, and CPU OPT values are working properly and agree with asus-wmi-sensors.

Great!

> * The three sensors showing REC2 don't work, I guess something else has to be changed.

They need to be read via other EC ports (see REC2() above).

zeule, 2022-02-10:
> I guess it is a conversion factor to volts. 0x0C35 (=3125) * 0.338 = 1.056, which is pretty close to 1.06.

Sorry, to µV, of course: 3125 * 338 * 1e-6 = 1.05625