zeule/asus-ec-sensors #16: Question: relevant for sTRX40 boards?

nabajour, 2022-02-09:
Hi,
I saw the article on this new driver on Phoronix. I have an Asus ROG Zenith II Extreme motherboard with a TRX40 chipset.
The current driver used is the `nct6789` driver. The information reported seems pretty generic (e.g., which fan info is the VRM fan?) and I'm not sure it's using the correct range scaling and offsets. 

Here is the `sensors` output:

```
nct6798-isa-0290
Adapter: ISA adapter
in0:                        1.17 V  (min =  +0.00 V, max =  +1.74 V)
in1:                      992.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in2:                        3.38 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in3:                        3.30 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in4:                        1.74 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in5:                      592.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in6:                        1.09 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in7:                        3.38 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in8:                        3.26 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in9:                      888.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in10:                     232.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in11:                     592.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in12:                       1.02 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in13:                       1.20 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in14:                       1.77 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
fan1:                      796 RPM  (min =    0 RPM)
fan2:                        0 RPM  (min =    0 RPM)
fan3:                      793 RPM  (min =    0 RPM)
fan6:                     2220 RPM  (min =    0 RPM)
fan7:                        0 RPM  (min =    0 RPM)
SYSTIN:                    +45.0°C  (high = +80.0°C, hyst = +75.0°C)  sensor = thermistor
CPUTIN:                    +48.5°C  (high = +80.0°C, hyst = +75.0°C)  sensor = thermistor
AUXTIN0:                   +21.5°C    sensor = thermistor
AUXTIN1:                   +72.0°C    sensor = thermistor
AUXTIN2:                   +16.0°C    sensor = thermistor
AUXTIN3:                   -20.0°C    sensor = thermistor
PECI Agent 0 Calibration:  +58.0°C
PCH_CHIP_CPU_MAX_TEMP:      +0.0°C
PCH_CHIP_TEMP:              +0.0°C
PCH_CPU_TEMP:               +0.0°C
intrusion0:               ALARM
intrusion1:               ALARM
beep_enable:              disabled
```

There seems to be a lot of activity around Asus sensors readout, between this repository, the WMI driver, the WMI EC driver. I'd be interested in having more detailed output for my board (and fan control if possible), but I'm not sure where to start. 

Decoding the DSDT doesn't show the BREC method (there are some ACPI errors on boot and on decode, though). But it seems to show the ASUSWMI device (AMW0), and a PNP0C09 device with methods that seem to do some EC Query, which seem to be relevant to this project. 

Any help to get the correct info and what project to talk to would be appreciated.

I can upload the DSDT dump if needed.

Thanks

nabajour, 2022-02-09:
The DSDT and SSDT dumps

[asus_rog_zenith_II_extreme_dsdt.zip](https://github.com/zeule/asus-ec-sensors/files/8032150/asus_rog_zenith_II_extreme_dsdt.zip)

zeule, 2022-02-09:
Hi,

this driver does not replace the nctXXX one, but rather complements it with other sensor reading. To replace generic names in sensors output you need to supply it with a config file that specifies those labels (see man sensors.conf). 

I briefly looked through the DSDT source code, and it differs from what I've seen before for X470, B550, and X570 chipets. Do you happen to run Windows on this machine? If so, maybe the [HWinfo](https://www.hwinfo.com/download/) app already knows is there anything to read from the EC on this board? It shows "ASUS EC" entry for it (and asks on startup whether you want to read EC sensors). 

Or you can just substitute your board name into one of the richest configuration in this driver (Crosshair VII Hero) and look at the sensors output. The mutex name is \_SB_.PCI0.SBRG.SIO1.MUT0 ("\\_SB_.PCI0.SBRG.SIO1.MUT0" as C char array. Please check the device name (part of the string up to .MUT0) in the dsdt source code (decompile as `iasl dsdt.dat`) should the driver is not able to find this mutex). Board name must match the content of /sys/class/dmi/id/board_name. Should not, probably, do any harm to the hardware.

Source code locations:
https://github.com/zeule/asus-ec-sensors/blob/master/asus-ec-sensors.c#L59
https://github.com/zeule/asus-ec-sensors/blob/master/asus-ec-sensors.c#L197

zeule, 2022-02-14:
For the record: yes, there are sensors exposed via the EC in this board.

nabajour, 2022-02-15:
Hi, 

I don't have any windows licence for this machine, so no luck to run HWINFO64. And it looks like the ASUS forums are pretty picky to let people access attachements, I have created an account, but I'm currently restricted to access it. I've seen some [screenshots of HWINFO64 on overclock.net](https://www.overclock.net/threads/asus-rog-zenith-ii-extreme-experience-discussion-thread.1737976/post-28681481), though. This [one seems to explicitly show the EC](https://www.overclock.net/threads/asus-rog-zenith-ii-extreme-experience-discussion-thread.1737976/post-28681808) interfaced information.

It took me a while to get to testing your suggestions and open the box to swap some fan connectors around to test the corresponding headers. I  did copy the CROSSHAIR VIII ZERO entry, added the VRM HS fan sensor, and changed the define for the mutex. 

Changes [here](https://github.com/nabajour/asus-ec-sensors/commit/adb87facfa11d9cc35e8a688eda20548028c7067).

For reference, I did this:
```
$ make
$ make modules
$ sudo make modules_install
```
On my Debian Sid, it was a bit unhappy about some stuff I don't understand enough about (didn't find `vmlinux`, complaining about BTF stuff). I added some symlinks and this seemed to compile the module and put it into `/usr/lib/modules/<uname -r>/extra/drivers/hwmon`. (I suspect Debian might store some files in different places)

It found the module at boot:
```
...

[    9.932993] AVX2 version of gcm_enc/dec engaged.
[    9.933246] AES CTR mode by8 optimization enabled
[    9.933401] thermal thermal_zone0: failed to read out thermal zone (-61)
[    9.945774] asus-ec-sensors PNP0C09:00: board has 13 EC sensors that span 18 registers
[    9.950852] Bluetooth: Core ver 2.22
[    9.950887] NET: Registered PF_BLUETOOTH protocol family
[    9.950889] Bluetooth: HCI device and connection manager initialized
[    9.950893] Bluetooth: HCI socket layer initialized
[    9.950895] Bluetooth: L2CAP socket layer initialized

...

```

Running `sensors` shows a new section at the top:

```
$ sensors
asusec-isa-0000
Adapter: ISA adapter
CPU Core:      1.01 V
CPU_Opt:        0 RPM
VRM HS:      8842 RPM
Chipset:     5289 RPM
Water_Flow:     0 RPM
Chipset:      +62.0°C
CPU:          +49.0°C
Motherboard:  +37.0°C
T_Sensor:     -40.0°C
VRM:          +39.0°C
Water_In:     -40.0°C
Water_Out:    -40.0°C
CPU:           6.00 A

nct6798-isa-0290
Adapter: ISA adapter
in0:                      992.00 mV (min =  +0.00 V, max =  +1.74 V)
in1:                      992.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in2:                        3.38 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in3:                        3.30 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in4:                        1.74 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in5:                      592.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in6:                        1.09 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in7:                        3.38 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in8:                        3.25 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in9:                      888.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in10:                     216.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in11:                     600.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in12:                       1.02 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in13:                       1.20 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in14:                       1.77 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
fan1:                      830 RPM  (min =    0 RPM)
fan2:                     2083 RPM  (min =    0 RPM)
fan3:                      894 RPM  (min =    0 RPM)
fan6:                        0 RPM  (min =    0 RPM)
fan7:                        0 RPM  (min =    0 RPM)
SYSTIN:                    +37.0°C  (high = +80.0°C, hyst = +75.0°C)  sensor = thermistor
CPUTIN:                    +44.0°C  (high = +80.0°C, hyst = +75.0°C)  sensor = thermistor
AUXTIN0:                   +21.5°C    sensor = thermistor
AUXTIN1:                   +80.0°C    sensor = thermistor
AUXTIN2:                   +16.0°C    sensor = thermistor
AUXTIN3:                   -20.0°C    sensor = thermistor
PECI Agent 0 Calibration:  +48.0°C
PCH_CHIP_CPU_MAX_TEMP:      +0.0°C
PCH_CHIP_TEMP:              +0.0°C
PCH_CPU_TEMP:               +0.0°C
intrusion0:               ALARM
intrusion1:               ALARM
beep_enable:              disabled

nvme-pci-2100
Adapter: PCI adapter
Composite:    +45.9°C  (low  = -273.1°C, high = +84.8°C)
                       (crit = +84.8°C)
Sensor 1:     +45.9°C  (low  = -273.1°C, high = +65261.8°C)
Sensor 2:     +47.9°C  (low  = -273.1°C, high = +65261.8°C)

iwlwifi_1-virtual-0
Adapter: Virtual device
temp1:        +49.0°C

k10temp-pci-00c3
Adapter: PCI adapter
Tctl:         +57.2°C
Tccd1:        +43.8°C
Tccd3:        +45.8°C
Tccd5:        +47.8°C
Tccd7:        +45.8°C

enp68s0-pci-4400
Adapter: PCI adapter
PHY Temperature:  +53.4°C
MAC Temperature:  +52.7°C

```
Fan speeds and temps seems to match the information in the BIOS better than the one from the NCT driver.

[image]
[image]
[image]

The "Water Flow" fan sensor seems to match "Fan 6" and is the water pump fan header on the motherboard. Didn't find what to match up the voltage and current information, though. 

So this looks promising! Anything more I should be looking at and test? Any more information we can eek out of this interface?

Thanks!

zeule, 2022-02-15:
Thank you for trying the driver out!

You can try to use DKMS (`make dkms` and `make dkms_clean` when installing a new version), another person with Debian wrote it works (#14).

Here is a fragment from the screenshot I've posted a link to:
[image]

It would be nice to find addresses for the missing temperature sensors (T_Sensor2 and EXT_Sensor2) and SB voltage. For the X470 boards the voltage was in registers 0x90 and 0x91, two-byte unsigned value which needs to be multiplied by 3125 to get µV, if I'm not mistaken. 

You can begin with dumping all EC registers using the ec_sys module:
```
# modprobe ec_sys
# hexdump -C /sys/kernel/debug/ec/ec0/io
```

Then we can look for additional blank temperature values (0xd8, i.e. -40) and fan speeds in the output.

nabajour, 2022-02-15:
Hi, thanks for the info on DKMS, I'll try this the next time I compile the module. It looks like I don't have that ec_sys module, it seems to be part of the debugging interface. 
From what I understand, I need to recompile my kernel with CONFIG_ACPI_EC_DEBUGFS=m. 
I'll come back with the info when I get around to recompile my kernel.

zeule, 2022-02-18:
@nabajour, could you show, please, board identification data: `grep ^ /sys/class/dmi/id/board_*`?

nabajour, 2022-02-21:
Finally got around to recompile my kernel, but it borked my bluetooth config along the way, needed to sort this out to have a cursor again before I could run these experiments X-D

> It would be nice to find addresses for the missing temperature sensors (T_Sensor2 and EXT_Sensor2) and SB voltage. For the X470 boards the voltage was in registers 0x90 and 0x91, two-byte unsigned value which needs to be multiplied by 3125 to get µV, if I'm not mistaken.
> 
> You can begin with dumping values from all EC registers int the bank 0 using the ec_sys module:
> 
> ```
> # modprobe ec_sys
> # hexdump -C /sys/kernel/debug/ec/ec0/io
> ```
> 
> Then we can look for additional blank temperature values (0xd8, i.e. -40) and fan speeds in the output.

```
$ sudo hexdump -C /sys/kernel/debug/ec/ec0/io        
00000000  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000030  00 00 00 00 00 00 00 00  00 00 3c 25 23 d8 2a 00  |..........<%#.*.|
00000040  7c 00 00 0f 0a 0f 0f ff  00 00 37 00 3c 66 5d ff  ||.........7.<f].|
00000050  00 00 00 00 00 ff ff 14  a9 00 05 00 00 00 00 00  |................|
00000060  1b 19 1f d8 d8 00 00 00  07 00 00 00 00 00 00 00  |................|
00000070  00 00 02 00 00 00 00 00  00 00 00 05 55 00 00 00  |............U...|
00000080  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
000000a0  00 00 03 e0 00 05 00 00  00 00 00 00 00 00 00 00  |................|
000000b0  00 00 00 00 14 9d 00 00  00 00 00 00 00 00 00 00  |................|
000000c0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
000000d0  40 10 10 41 00 00 00 00  34 2e 00 50 00 50 08 03  |@..A....4..P.P..|
000000e0  00 00 00 00 03 03 00 02  00 00 00 57 80 00 00 00  |...........W....|
000000f0  00 00 38 00 09 00 00 00  0a 0a 00 00 00 00 0c 00  |..8.............|
00000100
```

> @nabajour, could you show, please, board identification data: `grep ^ /sys/class/dmi/id/board_*`?

```
 $ sudo grep ^ /sys/class/dmi/id/board_*                 
/sys/class/dmi/id/board_asset_tag:Default string
/sys/class/dmi/id/board_name:ROG ZENITH II EXTREME
/sys/class/dmi/id/board_serial:XXXXXXXXXXXXXXXX
/sys/class/dmi/id/board_vendor:ASUSTeK COMPUTER INC.
/sys/class/dmi/id/board_version:Rev 1.xx
```
(dunno if I should post my board S/N here)

zeule, 2022-02-21:
Thanks for the update! Let's see... There are two oxd8 values (which is the blank value for temperature sensors) at 0x63 and 0x64. The water temperatures are, probably, in the second bank and can't be simply read by the ec_sys module.  Do not see any more obvious sensors. Maybe I should finally start disassemble UEFI code, where all these addresses should be present.

> (dunno if I should post my board S/N here)

Please don't. I'm sorry for forgetting about the serial field. I need only vendor, name, and version values.

zeule, 2022-07-08:
Resolved by #26.

closed 2022-07-08 by zeule

zeule, 2022-07-10:
@nabajour , I'm submitting these changes to the mainline kernel, and I would like to know your real email. Could you, please, send it directly to my email?