zeule/asus-ec-sensors #27: How does this driver compare to asus-wmi-sensors?

ajgringo619, 2022-08-16:
I'm asking for a specific reason: my Prime X470-Pro motherboard would infrequently lose all of its fans and shutdown with the aforementioned driver (https://github.com/electrified/asus-wmi-sensors), so when it became part of kernel 5.17 I had to blacklist it. Now that your driver is part of kernel 5.19, I'm a little leery of leaving this driver loaded.

Have you run across anything like this? I'm hoping because you're not using WMI that I'll be OK.

zeule, 2022-08-16:
This driver does not touch WMI functions, but instead reads from the ACPI embedded controller directly. The very reason for supporting the Prime X470-Pro model was to avoid lockups (see #12).

ajgringo619, 2022-08-16:
Perfect - thank you!

closed 2022-08-16 by ajgringo619

zeule, 2022-08-16:
My pleasure.