zeule/asus-ec-sensors #28: How to add support for ROG ZENITH EXTREME

arabek, 2022-09-15:
Currently the asus-wmi-sensors seems to be going to be deprecated in favour of the ec controller, which means, in the newer kernels i will loose info about what the mobo is doing.

```
asus_wmi_sensors-virtual-0
Adapter: Virtual device
CPU Core Voltage:             894.00 mV
CPU SOC Voltage:                1.09 V
DRAM AB Voltage:                1.20 V
DRAM CD Voltage:                1.20 V
1.8V PLL Voltage:               1.81 V
+12V Voltage:                  11.90 V
+5V Voltage:                    5.01 V
3VSB Voltage:                   3.31 V
VBAT Voltage:                   3.14 V
AVCC3 Voltage:                  3.29 V
SB 1.05V Voltage:               1.04 V
CPU Core Voltage:             849.00 mV
CPU SOC Voltage:                1.05 V
DRAM AB Voltage:                1.20 V
DRAM CD Voltage:                1.20 V
CPU Fan:                       712 RPM
Chassis Fan 1:                   0 RPM
Chassis Fan 2:                   0 RPM
HAMP Fan:                        0 RPM
Water Pump 1:                    0 RPM
CPU OPT:                         0 RPM
Water Flow:                      0 RPM
Waterblock Flow:                 0 RPM
EXT Fan 1:                       0 RPM
EXT Fan 2:                       0 RPM
EXT Fan 3:                       0 RPM
Cover Fan:                       0 RPM
CPU Temperature:               +36.0°C
CPU Socket Temperature:        +36.0°C
Motherboard Temperature:       +34.0°C
Chipset Temperature:           +52.0°C
Tsensor 1 Temperature:        +216.0°C
CPU VRM Temperature:           +39.0°C
Water In:                     +216.0°C
Water Out:                    +216.0°C
Waterblock In:                +216.0°C
Waterblock Out:               +216.0°C
EXT Tsensor 1:                  +0.0°C
EXT Tsensor 2:                  +0.0°C
EXT Tsensor 3:                  +0.0°C
Tsensor 2 Temperature:        +216.0°C
DIMM.2 Tsensor 1 Temperature: +216.0°C
DIMM.2 Tsensor 2 Temperature: +216.0°C
CPU VRM Output Current:         0.00 A
```
(with obvious errors around the ext tsensor and water block sensors, as i didn't bother to install them (nor do i have a watter cooling block, but the option is there), so the readings are wrong)

Is there a way to reliably add support for those sensors in the asus-ec-sensors module instead? I can hack my way around c code, so i'd do it myself, but need guidance.

zeule, 2022-09-15:
> Currently the asus-wmi-sensors seems to be going to be deprecated in favour of the ec controller…

That's probably not the case, these two drivers support different board families.

If you, however, want to contribute to this driver, you can copy the implementation for ROG ZENITH II EXTREME, I guess, and tune it. You need to provide DMI board identification (`grep -e '' -n /sys/class/dmi/id/*_name`) and the sensor set (which can be copied over from ZENITH II EXTREME).

zeule, 2022-09-16:
@arabek I updated the readme to contain up to date information on adding support for new boards.

arabek, 2022-10-03:
> That's probably not the case, these two drivers support different board families.

It's actually being dropped in kernel-6.1 via this pull req: https://lore.kernel.org/lkml/20221003000230.1008396-1-linux@roeck-us.net/

I've not yet had the time to work on support for the board, but will very soon. Thanks for the update to the readme.

zeule, 2022-10-03:
That's the old driver for ASUS EC, asus_wmi_ec_sensors, while you showed output from asus_wmi_sensors.

arabek, 2022-10-03:
Oh, okay. Silly me. In that case...

closed 2022-10-03 by arabek