zeule/asus-ec-sensors #29: Warning/Message about missing __x86_indirect_alt_call_rax when installing on 5.15 kernel

simonvanderveldt, 2022-09-18:
Hi Eugene,

Thanks for the work on this module! I just emerged the ebuild you linked in the readme and I got an warning message about `needs unknown symbol __x86_indirect_alt_call_rax`. I'm on the stable kernel in gentoo, so 5.15.63.
Modprobing it works fine though, and it also shows the sensor values in `sensors`, so not really sure if this is relevant?

I can't post the exact error unfortunately, since I cleared my terminal and for some reason re-emerging the module doesn't show the error anymore, even after unmerging it.

zeule, 2022-09-19:
Hey Simon, this might be caused by symbol trimming by the kernel. Is CONFIG_TRIM_UNUSED_KSYMS set?

simonvanderveldt, 2022-09-19:
Seems like it isn't no results for
```
zcat /proc/config.gz | grep -i CONFIG_TRIM_UNUSED_KSYMS
```
The gentoo-kernel kernels are using Fedora's kernel config, for 5.15.63 that's https://src.fedoraproject.org/rpms/kernel/raw/ec69da7a42b5b7c3da91572ef22097b069ddbd01/f/kernel-x86_64-fedora.config

zeule, 2022-09-19:
Help string for this option says it is enabled by default. I imagine the kernels trims symbols, but after you installed this module, its imports went into the list of used symbols in /lib/modules/${KVER}/build/ and now are guarded against trimming.

simonvanderveldt, 2022-09-19:
From all the checks I did it seems to be disabled, `make menuconfig` shows `N` as well (although the `.config` file is very different from the Fedora one, so not sure what's going on there).
Anyway, I guess it doesn't matter, either it's disabled or it's enabled and the symbol's now been added to the table so should be fine now?

zeule, 2025-10-08:
I hope the issue is not relevant anymore, and everyone updated to the latest kernel.

closed 2025-10-08 by zeule