zeule/asus-ec-sensors #32: Add note in README about upstreaming to kernel 6.1

h4emp3, 2023-01-21:
Hey there, I'm happily using this kernel module for a while now, very happy with it!

When I was updating to kernel 6.1 just now, dkms gave me an error telling me, that the module does already exist. So I guess you got this upstreamed? Congratulations!

I think it would be nice if this was readily appearent when reading the readme of this project though :)

zeule, 2023-01-21:
Thank you, apparently the message in the 3r paragraph of the readme is missable. Any suggestions?

h4emp3, 2023-01-21:
Well, damn. I guess my attention span isn't as great as I hoped :D  The paragraph is completely sufficient.

For what it's worth: I've searched for "upstream" and "6.1". But appearently 6.1 is not the point where this module was upstreamed, debian just did not prebuild/package it before this version. I did not even think about that possibility. And I also searched for some of the output dkms gives you when trying to build a kernel module that already exists.

Thanks for the quick reply and sorry for only skimming the readme and taking up your time because of it!

closed 2023-01-21 by h4emp3

zeule, 2023-01-21:
NP. Thanks for the clarification!