zeule/asus-ec-sensors #35: ASUS ProArt B550 Creator Support

fireflame90051, 2023-04-02:
Hey, I'm not sure if this is the best place to put this, but I needed readout of the AUX T_Sensor for a water-cooling project and did my best to add in support.

I found that many of the defined sensors report 0 or were duplicates of T_Sensor. i left the ones that did report and were not duplicates, along with the VRM sensor. On the subject of the VRM sensor, I don't believe this board has one as this driver reports 0 deg C, and it doesn't show up in the UEFI hardware monitor.

Btw, thanks for the descriptive documentation! I'm not exactly a programmer or a hardware guy but that made it super easy.

Here's a patch
```diff
--- asus-ec-orig/asus-ec-sensors.c	2023-04-02 05:57:52.493027229 -0500
+++ asus-ec-sensors/asus-ec-sensors.c	2023-04-02 05:38:50.992029670 -0500
@@ -303,6 +303,18 @@
 	.family = family_amd_500_series,
 };
 
+static const struct ec_board_info board_info_pro_art_b550_creator = {
+        .sensors = SENSOR_SET_TEMP_CHIPSET_CPU_MB | 
+
+                   //Driver reports 0 Deg C
+                   SENSOR_TEMP_VRM |
+ 
+                   SENSOR_TEMP_T_SENSOR |
+                   SENSOR_FAN_CPU_OPT,
+        .mutex_path = ASUS_HW_ACCESS_MUTEX_ASMX,
+        .family = family_amd_500_series,
+};
+
 static const struct ec_board_info board_info_pro_ws_x570_ace = {
 	.sensors = SENSOR_SET_TEMP_CHIPSET_CPU_MB | SENSOR_TEMP_VRM |
 		SENSOR_TEMP_T_SENSOR | SENSOR_FAN_CHIPSET |
@@ -435,6 +447,8 @@
 					&board_info_prime_x570_pro),
 	DMI_EXACT_MATCH_ASUS_BOARD_NAME("ProArt X570-CREATOR WIFI",
 					&board_info_pro_art_x570_creator_wifi),
+        DMI_EXACT_MATCH_ASUS_BOARD_NAME("ProArt B550-CREATOR",
+                                        &board_info_pro_art_b550_creator),
 	DMI_EXACT_MATCH_ASUS_BOARD_NAME("Pro WS X570-ACE",
 					&board_info_pro_ws_x570_ace),
 	DMI_EXACT_MATCH_ASUS_BOARD_NAME("ROG CROSSHAIR VIII DARK HERO",
```

Along with the output of asus-ec-sensors from sensors
```
❯ sensors asusec-isa-0000
asusec-isa-0000
Adapter: ISA adapter
CPU_Opt:      877 RPM
Chipset:      +67.0°C  
CPU:          +44.0°C  
Motherboard:  +41.0°C  
T_Sensor:     +25.0°C  
VRM:           +0.0°C  
```

zeule, 2023-04-02:
Hi, @fireflame90051, and thank you for the contribution! 

> … for a water-cooling project…

That's exactly how did I start this project.

If the VRM temp reading is always zero, I see no point in keeping the sensor. Maybe you'd like to submit a patch (with your name and email) so I could forward it to the mainline kernel? Otherwise I will just insert a link to this issue into the commit message.

40417256, 2023-04-02:
On my board the minimum temp is set to 60c in the bios so the fan will not start below. You can try to temporarily set it to a low temp and see if it start spinning (actually is there a VRM fan on your board?) 

If the VRM temp reading is always zero, I see no point in keeping the sensor. Maybe you'd like to submit a patch (with your name and email) so I could forward it to the mainline kernel? Otherwise I will just insert a link to this issue into the commit message.

zeule, 2023-04-02:
Excuse me?

fireflame90051, 2023-04-03:
> If the VRM temp reading is always zero, I see no point in keeping the sensor. Maybe you'd like to submit a patch (with your name and email) so I could forward it to the mainline kernel? Otherwise I will just insert a link to this issue into the commit message.

The VRM sensor has been removed. 

Please excuse my ignorance. I'm pretty new to github, as this is my first contribution I'm not entirely sure how you want me to submit that patch.  Ive attached the zipped patch file here (github wouldn't allow me to attach the .patch file directly for whatever reason), and also created a fork on my account that we could merge if you want.

This patch was compiled against the current master branch and installed via dkms just before uploading so there shouldn't be any issues.

Thanks!

[b550-creator-patch.zip](https://github.com/zeule/asus-ec-sensors/files/11135612/b550-creator-patch.zip)

zeule, 2023-04-03:
Thank you for updating the patch! 

> I'm not entirely sure how you want me to submit that patch

Ideally, I'd like to get the patch (that is already here), your consent to submit the change to the mainline kernel under the GPL license, and your email to put in the patch for mainlining. Whether you use GitHub pull request or another way to share the code change is of no importance for me as far as I can read the changes. If you do not want to share your email on GitHub, but want it to be in the mainline kernel commit, please send an email to my address, which is in the GitHub profile.

fireflame90051, 2023-04-03:
Oh cool. my email is cacoukoulis@gmail.com, I also left my email and name as a comment in the patch file itself.

You absolutely have my consent. Personally, I think it's pretty cool that my teeny tiny contribution may end up being in mainline :)

closed 2023-04-03 by fireflame90051