zeule/asus-ec-sensors #38: looking to add sensors support for TP3402ZA

ntzb, 2023-04-16:
Hi,
I have a laptop with fans that work too hard. using lm-sensors produce zero fan/voltage info.
during my searches I found that the laptop has an ITE chip IT5571 which is EC.
I'm currently trying to get SOME sensor reading information.
I have Windows in dual boot, but HWiNFO/speedfan do not show any fan/voltage info.

trying to work under the readme, in HWiNFO I do not see `ASUS EC`.
what I do have is: `Super-IO/LPC Chip: ITE IT5571` (full dump at https://pastebin.com/hfK4tR2C)
and this info:
```
/sys/class/dmi/id/board_name:1:TP3402ZA
/sys/class/dmi/id/board_vendor:1:ASUSTeK COMPUTER INC.
```
the `dsdt.dsl` file is attached ([dsdt.txt](https://github.com/zeule/asus-ec-sensors/files/11243007/dsdt.txt)), but I didn't really understand how to find an address or extract the needed information from it.
any help would be really appreciated!

zeule, 2023-04-16:
Hi,
from the first glance the DSDT code is very different from what I've seen for motherboards, supported by this driver.

ntzb, 2023-04-16:
does this mean that no sensor information whatsoever can be extracted for this laptop?
ASUS has MyASUS app on Windows that can read the fan information, and control 3 states (silent, regular, performance).
unfortunately, lm-sensors can't add any sensor information for IT5571, and the only reference I saw for ASUS EC is your repo.

zeule, 2023-04-16:
I'm saying that this driver is not going to work with the laptop. Although I see something that might be the fan control method, it is very different from what ASUS put in their desktop boards. 

> can read the fan information, and control 3 states (silent, regular, performance).

The FANL function in the DSDT dump (exported via WMI) seems to be doing exactly that. You can try to call it using WMI or using direct ACPI call (there is acpi_call kernel module that allows to perform ACPI call from userspace).

ntzb, 2023-04-16:
> I'm saying that this driver is not going to work with the laptop. Although I see something that might be the fan control method, it is very different from what ASUS put in their desktop boards.
> 
this is unfortunate
> The FANL function in the DSDT dump (exported via WMI) seems to be doing exactly that. You can try to call it using WMI or using direct ACPI call (there is acpi_call kernel module that allows to perform ACPI call from userspace).
>
but this is nice, and it works! thanks so much!!
built and installed acpi_call from https://github.com/nix-community/acpi_call, loaded the .ko (`insmod acpi_call.ko`) and ran:
`echo "\_SB.ATKD.FANL" 0x1 > /proc/acpi/call` and the fan is almost silent.
it seems 0x1 is silent mode, but there's not much difference between 0x0 and 0x2 (but I guess it's safe to assume 0x0 is the default no?).
this looks interesting:
```
    External (_SB_.PC00.LPCB.H_EC.BAT0, DeviceObj)
    External (_SB_.PC00.LPCB.H_EC.BAT1, DeviceObj)
    External (_SB_.PC00.LPCB.H_EC.BAT2, DeviceObj)
    External (_SB_.PC00.LPCB.H_EC.BNUM, IntObj)
    External (_SB_.PC00.LPCB.H_EC.CFAN, FieldUnitObj)
    External (_SB_.PC00.LPCB.H_EC.CHRG, DeviceObj)
    External (_SB_.PC00.LPCB.H_EC.ECMD, MethodObj)    // 1 Arguments
    External (_SB_.PC00.LPCB.H_EC.ECNT, MethodObj)    // 1 Arguments
    External (_SB_.PC00.LPCB.H_EC.ECRD, MethodObj)    // 1 Arguments
    External (_SB_.PC00.LPCB.H_EC.ECWT, MethodObj)    // 2 Arguments
    External (_SB_.PC00.LPCB.H_EC.LID0, DeviceObj)
    External (_SB_.PC00.LPCB.H_EC.LSTE, FieldUnitObj)
    External (_SB_.PC00.LPCB.H_EC.ONTM, IntObj)
    External (_SB_.PC00.LPCB.H_EC.RPWR, FieldUnitObj)
    External (_SB_.PC00.LPCB.H_EC.SKEY, MethodObj)    // 1 Arguments
    External (_SB_.PC00.LPCB.H_EC.UPBT, MethodObj)    // 2 Arguments
    External (_SB_.PC00.LPCB.H_EC.UTEC, MethodObj)    // 2 Arguments
    External (_SB_.PC00.LPCB.H_EC.VPWR, FieldUnitObj)
    External (_SB_.PC00.LPCB.H_EC.WTMM, FieldUnitObj)
    External (_SB_.PC00.LPCB.H_EC.WTMV, FieldUnitObj)
    External (_SB_.PC00.LPCB.H_EC.XDAT, MethodObj)    // 0 Arguments
```

I wonder if `CFAN` holds the RPM (or PWM) data, and if we can somehow view it.

zeule, 2023-04-16:
Glad it works! You can decompile the WMI data (MOF in WQMO and something else in_WDG ones, you can find information online) and that might give hints. Otherwise just read the decompiled DSDT and guess…

I'm closing the issue, but I'll be happy to help you answering questions here.

closed 2023-04-16 by zeule