zeule/asus-ec-sensors #41: Add support for x470-f

whatsitsname, 2023-04-30:
Hello,  I believe x470-f may be supported by this driver.  I exhibit similar behavior when using  asus-wmi-sensor as mentioned in issue #12.

PNP0C09 show up in dsdt.   However; when I try to add support for x470-f  loading the modules if fails.

sudo modprobe -v  asus-ec-sensors
insmod /lib/modules/6.2.12-1-default/kernel/drivers/hwmon/asus-ec-sensors.ko.zst 
modprobe: ERROR: could not insert 'asus_ec_sensors': No such device

[dsdt.dsl.txt](https://github.com/zeule/asus-ec-sensors/files/11361569/dsdt.dsl.txt)

/sys/class/dmi/id/board_name:1:ROG STRIX X470-F GAMING
/sys/class/dmi/id/board_vendor:1:ASUSTeK COMPUTER INC.

Let know how I can help you.
Perhaps there is hope this board will be supported someday.

I am using the latest bios (6063).

I appreciate your time.

zeule, 2023-04-30:
Hello! Sorry, I could not understand from your message, have you tried to make necessary changes to the driver, and it fails to load, or did you simply tried to load it as is? From the DSDT the board looks very similar to X470 PRO, thus we can try a similar set of sensors.

whatsitsname, 2023-04-30:
I did in fact try to make the changes......my modification were that I added 

	DMI_EXACT_MATCH_ASUS_BOARD_NAME("ROG STRIX X470-F GAMING",
					&board_info_strix_x470_f),

and 

static const struct ec_board_info board_info_strix_x470_f = {
	.sensors = SENSOR_SET_TEMP_CHIPSET_CPU_MB |
		SENSOR_TEMP_T_SENSOR | SENSOR_TEMP_VRM |
		SENSOR_FAN_CPU_OPT |
		SENSOR_CURR_CPU | SENSOR_IN_CPU_CORE,
	.mutex_path = ACPI_GLOBAL_LOCK_PSEUDO_PATH,
	.family = family_amd_400_series,
};

Perhaps I missed something.
I then looked for advice from yourself.

zeule, 2023-04-30:
That looks correct. 'modprobe: ERROR: could not insert 'asus_ec_sensors': No such device' means the board name was not found in the dmi_table array. 

Are you sure the changes you've made reached /lib/modules/6.2.12-1-default/kernel/drivers/hwmon/asus-ec-sensors.ko.zst. I suggest you to clone this repository, apply changes there,  call `make` and use `sudo insmod ./asus-ec-sensors.ko` to ensure you are inserting the code you work on.

whatsitsname, 2023-04-30:
your right it worked

sensors

asus_wmi_sensors-virtual-0
Adapter: Virtual device
CPU Core Voltage:        894.00 mV 
+12V Voltage:             12.29 V  
+5V Voltage:               5.01 V  
3VSB Voltage:              3.36 V  
CPU Fan:                  410 RPM
Chassis Fan 1:           1015 RPM
Chassis Fan 2:           1018 RPM
Chassis Fan 3:           1021 RPM
AIO Pump:                   0 RPM
Water Pump:                 0 RPM
CPU OPT:                  432 RPM
CPU Temperature:          +23.0°C  
Motherboard Temperature:  +29.0°C  
Chipset Temperature:      +43.0°C  
Tsensor 1 Temperature:   +216.0°C  

asusec-isa-0000
Adapter: ISA adapter
CPU Core:      1.47 V  
CPU_Opt:      432 RPM
Chipset:      +43.0°C  
CPU:          +23.0°C  
Motherboard:  +29.0°C  
T_Sensor:     -40.0°C  
VRM:          +22.0°C  
CPU:           5.00 A  

k10temp-pci-00c3
Adapter: PCI adapter
Tctl:         +24.0°C  

I'm thinking the MB struct needs some work.  ?

zeule, 2023-04-30:
Glad you succeeded! What looks incomplete to you regarding the sensors definition?

whatsitsname, 2023-04-30:
If you look as my reply and compare the output of  asus-wmi-sensor  to asus-ec-sensors  things start to not line up.

I'm only seeing the fan speed for one of the two cpu fan headers.  Also the list of shown sensors is a bit shorter than the wmi  sensor.   IE... the chasis fans ;   water pump headers.   12v 5v 3.3v    CPU current i've never seen before!

zeule, 2023-04-30:
In these ASUS boards the monitoring info is split between the Super I/O chip and the Embedded Controller. The EC registers are used for non-tunable ones, like temperatures or voltages, or fans without user-tunable qurve. I'm unsure why is the CPU_Opt fan among those.

whatsitsname, 2023-04-30:
Are you suggesting the x470-f  is not so easily implemented? 
Or perhaps having them both loaded is causing issues?

zeule, 2023-04-30:
There are two chips related to hardware monitoring in these boards: the Super I/O one and the EC. Back in the days there was only one, the SIO one, but the number of sensors grows and ASUS moved some of them to the EC.  To get reading for the rest of the sensors published by asuswmi, you need to find out which Super I/O chip is installed in the board and use hwmon driver for it. If this sensor does not show up automatically, you might want to check out [kernel bug 204807](https://bugzilla.kernel.org/show_bug.cgi?id=204807), it mensions X470-F GAMING.

whatsitsname, 2023-04-30:
I find that bug report confusing at best the x470-f  has a IT8665e not a nct6775.
I will be patient.  Thank you for your time.

zeule, 2023-04-30:
Sorry, I just remember the board was mentioned there, but I have never went into details. IT8665e is a sad story in Linux unfortunately.

whatsitsname, 2023-04-30:
Well it was worth a poke.   DOH!