zeule/asus-ec-sensors #45: Add support for ROG STRIX X670E-E GAMING WIFI

ZVNexus, 2023-10-19:


sevens, 2023-11-22:
I have the same motherboard.  Tested https://github.com/ZVNexus/asus-ec-sensors/commit/bd4e4fed2a7b40e689e8ed4faec3ab1bc87b0d2e and got some output as well:

```
# sensors 'asusec-*' 'k10temp-*'
asusec-isa-0000
Adapter: ISA adapter
CPU:          +31.0°C
CPU Package:  +37.0°C
Motherboard:  +31.0°C
VRM:          +34.0°C
Water_In:     +35.0°C
Water_Out:    +33.0°C

k10temp-pci-00c3
Adapter: PCI adapter
Tctl:         +37.6°C
Tccd1:        +30.2°C
Tccd2:        +29.5°C
```

Note that I don't have a water cooler connected so not sure why the water in and out give any values (nor if this comes from the hardware or software).  FWIW, loading the `nct6775` module as well and checking `sensors` as a quick check gives no similar temperatures to the water ones (ignoring decimals).

<details>
<summary>With nct6799 output</summary>

```
# sensors 'asusec-*' 'k10temp-*' 'nct6799-*'
asusec-isa-0000
Adapter: ISA adapter
CPU:          +31.0°C
CPU Package:  +37.0°C
Motherboard:  +30.0°C
VRM:          +34.0°C
Water_In:     +36.0°C
Water_Out:    +33.0°C

k10temp-pci-00c3
Adapter: PCI adapter
Tctl:         +37.5°C
Tccd1:        +29.9°C
Tccd2:        +29.5°C

nct6799-isa-0290
Adapter: ISA adapter
in0:                      360.00 mV (min =  +0.00 V, max =  +1.74 V)
in1:                      1000.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in2:                        3.38 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in3:                        3.31 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in4:                        1.02 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in5:                      992.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in6:                        1.10 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in7:                        3.38 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in8:                        3.28 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in9:                        3.31 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in10:                       1.06 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in11:                       1.06 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in12:                       1.03 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in13:                       1.02 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in14:                       1.22 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
fan1:                        0 RPM  (min =    0 RPM)
fan2:                      368 RPM  (min =    0 RPM)
fan3:                        0 RPM  (min =    0 RPM)
fan4:                      237 RPM  (min =    0 RPM)
fan5:                      233 RPM  (min =    0 RPM)
fan6:                      244 RPM  (min =    0 RPM)
fan7:                        0 RPM  (min =    0 RPM)
SYSTIN:                    +30.0°C  (high = +80.0°C, hyst = +75.0°C)  sensor = thermistor
CPUTIN:                    +31.0°C  (high = +80.0°C, hyst = +75.0°C)  sensor = thermistor
AUXTIN0:                   +20.0°C    sensor = thermistor
AUXTIN1:                   +23.0°C    sensor = thermistor
AUXTIN2:                   +23.0°C    sensor = thermistor
AUXTIN3:                   +14.0°C    sensor = thermistor
PECI Agent 0 Calibration:  +31.0°C
PCH_CHIP_CPU_MAX_TEMP:      +0.0°C
PCH_CHIP_TEMP:              +0.0°C
PCH_CPU_TEMP:               +0.0°C
TSI0_TEMP:                 +37.4°C
intrusion0:               OK
intrusion1:               ALARM
beep_enable:              disabled
```

</details>

This is with lm-sensors 3.6.0 and kernel 6.5.12 (Slackware64-current).

Let me know if there's anything further I can do to help/test (I don't have Windows available).

zeule, 2023-11-22:
Thank you for testing this out! So, water temperatures are likely invalid and display something else.

sevens, 2023-12-24:
I recently did some more testing with an AIO watercooler connected (pump and fans connected, cooling block not mounted to the CPU as it was just a short test of the AIO device itself, just ran some extra commands in the OS while it was connected).  I didn't notice any relevant difference in the output with it connected:

```
#  sensors 'asusec-*'
asusec-isa-0000
Adapter: ISA adapter
CPU:          +28.0°C
CPU Package:  +37.0°C
Motherboard:  +28.0°C
VRM:          +31.0°C
Water_In:     +36.0°C
Water_Out:    +33.0°C
```

Investigating a bit further I couldn't remember reading anything about water temperature sensors when reading the manual nor seeing any connectors for it, let alone 2, on the actual motherboard.  Searching [the manual](https://dlcdnets.asus.com/pub/ASUS/mb/Socket%20AM5/ROG_STRIX_X670E-E_GAMING_WIFI/E21210_ROG_STRIX_X670E-E_GAMING_WIFI_UM_V3_WEB.pdf?model=ROG%20STRIX%20X670E-E%20GAMING%20WIFI) again for 'water' gives no relevant results (only one match, about where to connect the AIO pump header to).  Searching for 'temper' (for temperature) also had no relevant hits.  Though there is a single generic header for a temperature sensor but AFAICT that's for a generic sensor to place somewhere in your case (and e.g. let the EFI use this for the fan curves).

Searching in the EFI itself (`F9`) for 'water' also yields no results.

So in short: I agree they're likely displaying something else (though no idea what) :P

This was with kernel version 6.6.3.

P.s.: there is also a "[ROG STRIX X670E Series BIOS Manual](https://dlcdnets.asus.com/pub/ASUS/mb/Socket%20AM5/ROG_STRIX_X670E-E_GAMING_WIFI/E20544_ROG_STRIX_X670E_Series_BIOS_manual_EM_WEB.pdf?model=ROG%20STRIX%20X670E-E%20GAMING%20WIFI)", which, judging by the name, is for the entire STRIX X670E series (there are quite some variants).  This does mention water in and out sensors multiple times (e.g. in the 'Temperature Monitor' and 'Q-Fan Source').

I compared with the 'Temperature Monitor' page in the EFI on my motherboard and there are no water sensors there.  This is without the AIO connected again, but there are multiple other sensors that aren't connected either but are still listed, e.g. the generic "T_Sensor", and DIMM temperatures for the 2 unpopulated slots.

Also checked the 'Q-Fan Source' configuration and the only options there are: `CPU`, `CPU Package`, `MotherBoard`, `VRM`, `T_Sensor` and `Multiple sources` (using the latter gives multiple drop-downs for sources, each invidual containing the same options as mentioned here, sans `Multiple sources` of course).  Note that this is for the chassis fans, the CPU fan only has `CPU` and `CPU Package`.

zeule, 2023-12-25:
There should be two-pin connectors for water temperature sensors on the motherboard, if they are absent there is no support for water temperature. To merge in the changes from #46, those entries need to be removed.

ZVNexus, 2023-12-26:
@zeule Sorry for lack of updates, I no longer own the board and switched to an ASRock X670E Taichi. Board has some ACPI bugs that ASUS has shown a disinterest in fixing which makes it unsuitable for use on Linux.

zwostein, 2023-12-27:
Found the `T_Sensor` register. Works fine with a thermistor connected to the "Thermal Sensor" header on the motherboard.

```
asusec-isa-0000
Adapter: ISA adapter
CPU:          +37.0°C  
CPU Package:  +45.0°C  
Motherboard:  +28.0°C  
T_Sensor:     +30.0°C  
VRM:          +40.0°C
```

sevens, 2023-12-31:
@ZVNexus : if you don't mind me asking, what ACPI issues did you have?  I couldn't find anything with a quick search.  Might be relevant to me and others here as well ;)

And thanks for updating the patch anyway.

ZVNexus, 2024-01-02:
> @ZVNexus : if you don't mind me asking, what ACPI issues did you have? I couldn't find anything with a quick search. Might be relevant to me and others here as well ;)
> 
> And thanks for updating the patch anyway.

https://www.reddit.com/r/buildapc/comments/xypn1m/network_card_intel_ethernet_controller_i225v_igc/

This is a tremendous oversight from ASUS and when I reached out to them, they were not interested in fixing their firmware bug at all. I looked at the ACPI tables and they're missing the entries for I225. The Crosshair board doesn't suffer from the same issue (or any other boards in the X670/E lineup that I could see).

Just went out and bought the X670E Taichi instead of trying to workaround something that should have never been an issue. Interesting bit about the board, it doesn't have *any* of the typical "ACPI BIOS BUG" dmesg warnings either. I had a chance to interact with ASRock's support a bit ago and they actually test Linux from time to time on their boards, and fix issues they observe.

IMO ASRock is probably the best choice if you're looking for a new board at some point.

sevens, 2024-01-04:
> > @ZVNexus : if you don't mind me asking, what ACPI issues did you have? I couldn't find anything with a quick search. Might be relevant to me and others here as well ;)
> > And thanks for updating the patch anyway.
> 
> https://www.reddit.com/r/buildapc/comments/xypn1m/network_card_intel_ethernet_controller_i225v_igc/
> 
> This is a tremendous oversight from ASUS and when I reached out to them, they were not interested in fixing their firmware bug at all.
> 
> Just went out and bought the X670E Taichi instead of trying to workaround something that should have never been an issue. Interesting bit about the board, it doesn't have _any_ of the typical "ACPI BIOS BUG" dmesg warnings either. I had a chance to interact with ASRock's support a bit ago and they actually test Linux from time to time on their boards, and fix issues they observe.
> 
> IMO ASRock is probably the best choice if you're looking for a new board at some point.

Thanks!  Haven't had the issue myself but I'll keep an eye on it (have had it running for many hours in a row, with wired only network but no heavy load on the PC/networking yet; I do have all settings that the thread says should cause the problem though).

Good to know about ASRock's Linux testing as well.

sevens, 2024-03-20:
I finally got around to test [cb31069](https://github.com/ZVNexus/asus-ec-sensors/commit/cb31069f2a561b3872be0b47db597b08f6945eb6).  This version seems to work fine as well:

```sh
#  sensors 'asusec-*' 'k10temp-*' 'nct6799-*' -c /dev/null
asusec-isa-0000
Adapter: ISA adapter
CPU:          +37.0°C
CPU Package:  +43.0°C
Motherboard:  +36.0°C
T_Sensor:     -60.0°C
VRM:          +40.0°C

k10temp-pci-00c3
Adapter: PCI adapter
Tctl:         +43.6°C
Tccd1:        +35.5°C
Tccd2:        +34.6°C

nct6799-isa-0290
Adapter: ISA adapter
...
AUXTIN5:                       -60.0°C
...
```

Note that I don't have anything connected to `T_Sensor` nor anything I can connect to it for testing.  Its value seems to match `AUXTIN5` from the `nct6799` module in my case but that might be a coincidence (e.g. both showing some sort of default).

Tested on kernel 6.6.22.

<details>
  <summary>full nct6799 output (for reference)</summary>

```
nct6799-isa-0290
Adapter: ISA adapter
in0:                          920.00 mV (min =  +0.00 V, max =  +1.74 V)
in1:                          1000.00 mV (min =  +0.97 V, max =  +1.03 V)
in2:                            3.38 V  (min =  +3.28 V, max =  +3.49 V)
in3:                            3.31 V  (min =  +3.22 V, max =  +3.41 V)
in4:                          1000.00 mV (min =  +0.97 V, max =  +1.03 V)
in5:                          992.00 mV (min =  +0.92 V, max =  +1.08 V)
in6:                            1.10 V  (min =  +1.06 V, max =  +1.14 V)
in7:                            3.38 V  (min =  +3.28 V, max =  +3.49 V)
in8:                            3.30 V  (min =  +3.18 V, max =  +3.38 V)
in9:                            3.31 V  (min =  +3.22 V, max =  +3.41 V)
in10:                           1.06 V  (min =  +1.02 V, max =  +1.09 V)
in11:                           1.06 V  (min =  +1.02 V, max =  +1.09 V)
in12:                           1.03 V  (min =  +1.00 V, max =  +1.06 V)
in13:                           1.02 V  (min =  +0.99 V, max =  +1.05 V)
in14:                           1.22 V  (min =  +1.18 V, max =  +1.26 V)
in15:                         808.00 mV (min =  +0.74 V, max =  +0.90 V)
in16:                           3.38 V  (min =  +3.28 V, max =  +3.49 V)
in17:                           2.03 V  (min =  +1.97 V, max =  +2.04 V)
fan1:                            0 RPM  (min =    0 RPM)
fan2:                          438 RPM  (min =    0 RPM)
fan3:                            0 RPM  (min =    0 RPM)
fan4:                          315 RPM  (min =    0 RPM)
fan5:                          319 RPM  (min =    0 RPM)
fan6:                          323 RPM  (min =    0 RPM)
fan7:                            0 RPM  (min =    0 RPM)
SYSTIN:                        +36.0°C  (high = +80.0°C, hyst = +75.0°C)
                                        (crit = +125.0°C)  sensor = thermistor
CPUTIN:                        +37.0°C  (high = +80.0°C, hyst = +75.0°C)
                                        (crit = +125.0°C)  sensor = thermistor
AUXTIN0:                       +21.0°C  (high = +80.0°C, hyst = +75.0°C)
                                        (crit = +125.0°C)  sensor = thermistor
AUXTIN1:                       +23.0°C  (high = +80.0°C, hyst = +75.0°C)
                                        (crit = +125.0°C)  sensor = thermistor
AUXTIN2:                       +23.0°C  (high = +80.0°C, hyst = +75.0°C)
                                        (crit = +125.0°C)  sensor = thermistor
AUXTIN3:                       +14.0°C  (high = +80.0°C, hyst = +75.0°C)
                                        (crit = +125.0°C)  sensor = thermistor
AUXTIN4:                       +26.0°C  (high = +80.0°C, hyst = +75.0°C)
                                        (crit = +125.0°C)
PECI/TSI Agent 0 Calibration:  +37.0°C  (high = +80.0°C, hyst = +75.0°C)
AUXTIN5:                       -60.0°C
PCH_CHIP_CPU_MAX_TEMP:          +0.0°C
PCH_CHIP_TEMP:                  +0.0°C
PCH_CPU_TEMP:                   +0.0°C
TSI0_TEMP:                     +43.5°C
intrusion0:                   OK
intrusion1:                   ALARM
beep_enable:                  disabled
```

</details>

sevens, 2024-04-24:
Unfortunately my C development skills are at the "just might get a hello world to work" level.

ZVNexus, 2024-04-24:
I will look into submitting the changes to the mailing list Friday, or you can submit them if you would like it sooner (not sure around the cut off dates for making it into 6.10).

zeule, 2024-04-25:
Oh, thank you! Maybe you can comment on issue #43, please?

sevens, 2024-05-21:
> @ZVNexus , would you like to prepare the submission? If you do not want to (given you don't use the board anymore), may I submit your changes, please? @sevens, would you agree to test?

Sorry for the late reply, I missed that I got mentioned in there...  No problem to test anything (if it's still needed)!

Just upgraded my kernel to 6.9.1 this week and it's still working fine in there.

<details>

<summary>kernel 6.9.1 (room temperature: 22.9°C)</summary>

```
#  sensors 'asusec-*' 'k10temp-*' 'nct6799-*' -c /dev/null
asusec-isa-0000
Adapter: ISA adapter
CPU:          +34.0°C
CPU Package:  +40.0°C
Motherboard:  +34.0°C
T_Sensor:     -60.0°C
VRM:          +37.0°C

k10temp-pci-00c3
Adapter: PCI adapter
Tctl:         +40.9°C
Tccd1:        +34.1°C
Tccd2:        +33.2°C

nct6799-isa-0290
Adapter: ISA adapter
...
AUXTIN5:                       -60.0°C
...
```

</details>