zeule/asus-ec-sensors #49: add uninstall instructions/functionality

realies, 2024-03-17:


Thorazin, 2024-08-04:
Yeah...  This is what I get when I run the "install instructions"

jason@LinuxBox:~/Downloads/asus-ec-sensors$ sudo make dkms
[sudo] password for jason: 
make: dkms: No such file or directory
make: *** [Makefile:56: dkms] Error 127

Thorazin, 2024-08-04:
They need to be more thorough and complete.

Thorazin, 2024-08-04:
and then there is this:

jason@LinuxBox:~/Downloads/asus-ec-sensors$ sudo make
[sudo] password for jason: 
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc-13 (Ubuntu 13.2.0-23ubuntu4) 13.2.0
  You are using:           
  CC [M]  /home/jason/Downloads/asus-ec-sensors/asus-ec-sensors.o
/bin/sh: 1: gcc-13: not found
make[2]: *** [scripts/Makefile.build:243: /home/jason/Downloads/asus-ec-sensors/asus-ec-sensors.o] Error 127
make[1]: *** [Makefile:1926: /home/jason/Downloads/asus-ec-sensors] Error 2
make: *** [Makefile:43: modules] Error 2

KeithMyers, 2024-08-04:
You need to bring your gcc up to the same version as the kernel was built with.

sudo apt install gcc-13 g++13

sudo update-alternatives --install /usr/bin/gcc gcc /usr/bin/gcc-13 13 --slave /usr/bin/g++ g++ /usr/bin/g++-13

I just had to do this to move to 6.10.3 which is built with gcc-14 and have asus-ec-sensors install with sudo make dkms but of course my update-alternatives command was for gcc-14