zeule/asus-ec-sensors #50: Will the ASUS ProArt X670E be included?

KeithMyers, 2024-04-24:
The next board I purchase for my daily driver will be the ASUS ProArt X670E motherboard.  Will it be included in the driver?

It has all the features that I need and is not as expensive as my ASUS Crosshair X670E Hero motherboards which are included in the driver.

Would be a shame to lose monitoring on this new board.

zeule, 2024-04-25:
Only those with access to the hardware are able to build support for it.

KeithMyers, 2024-04-25:
I don't understand your response.  Are you saying you won't be supporting the board unless you have the board yourself?

KeithMyers, 2024-04-25:
So no existing user of the board has approached you yet?  I find that hard to believe.

zeule, 2024-04-25:
Well, if the board is not on list of supported ones, and there is no PR for the model, the conclusion is clear: nobody submitted the support for that board yet.

zeule, 2024-04-25:
It does not necessary mean he wants to spend hours locating sensor readings in hex dumps...

KeithMyers, 2024-04-25:
I just asked the community for assistance on the board in the review comments.

KeithMyers, 2024-05-03:
Just got the ASUS ProArt X670E motherboard and RAM today.  But the cpu won't be back in stock for another couple or weeks.  Then I can get started with the acpidump and DSDT files.

KeithMyers, 2024-05-25:
Just added my board to the .C file
Built and installed just fine.
ASUS ProArt X670E-CREATOR WIFI

Here are the DSDT.DAT and DSDT.DSL files
[dsdt.zip](https://github.com/zeule/asus-ec-sensors/files/15445116/dsdt.zip)

Here is the output:

asusec-isa-0000
Adapter: ISA adapter
CPU:          +77.0°C  
CPU Package:  +88.0°C  
Motherboard:  +32.0°C  
VRM:          +53.0°C  
Water_In:     +36.0°C  
Water_Out:    +33.0°C 

Just needed to copy the Crosshair X670E structure and change the naming to suit the ProArt X670E-CREATOR WIFI board name.

static const struct ec_board_info board_info_pro_art_x670E_creator = {
	.sensors = SENSOR_TEMP_CPU | SENSOR_TEMP_CPU_PACKAGE |
		SENSOR_TEMP_MB | SENSOR_TEMP_VRM |
		SENSOR_SET_TEMP_WATER,
	.mutex_path = ACPI_GLOBAL_LOCK_PSEUDO_PATH,
	.family = family_amd_600_series,
};

	DMI_EXACT_MATCH_ASUS_BOARD_NAME("ProArt X670E-CREATOR WIFI",
					&board_info_pro_art_x670E_creator),

You should add this board to the compatible list now.

KeithMyers, 2024-05-25:
Thought it should be easy.

closed 2024-05-25 by KeithMyers

zeule, 2024-05-26:
Excellent, than you!

KeithMyers, 2024-05-26:
Needed to fix up a bit after I realized that  there were not Water In and Water Out headers and instead just a T_sensor header.

Wasted a couple hours trying to find the correct register address for that header.  Then this morning finally looked at the [ROG STRIX X670E-E GAMING WIFI issue and found that someone had already found the T_sensor header.

So changed to:
[ec_sensor_temp_t_sensor] =
		EC_SENSOR("T_Sensor", hwmon_temp, 1, 0x00, 0x36)
and recompiled.

Now the T_sensor output is closer to what the inline temp sensor/flowmeter is showing.

To prove it actually is the correct address, pulled the T_sensor thermistor header and the output changed to -62 which proves that is the correct address when the header goes unterminated.

KeithMyers, 2024-05-26:
Your commit https://github.com/ZVNexus/asus-ec-sensors/commit/cb31069f2a561b3872be0b47db597b08f6945eb6 is correct for the ProArt X670E-CREATOR WIFI also with just the board name needing to be updated.