zeule/asus-ec-sensors #51: DSDT ASUS ROG STRIX Z370-E Gaming

FabXxXxXx, 2024-05-03:
Hi,

This is the DSDT file for my good old ASUS ROG STRIX Z370-E Gaming. Is it possible to add support for it in LibreHardwareMonitorLib ?

Thank you

[DSDT.zip](https://github.com/zeule/asus-ec-sensors/files/15195440/DSDT.zip)

zeule, 2024-05-04:
Hi,

thank you for the DSDT dump. It would be more useful in the binary format (not decompiled), though... Excuse me, I don't really understand what are you asking for. This is a repository for Linux HWMON driver, why are you asking to add support to LibreHardwareMonitor here?

FabXxXxXx, 2024-05-06:
Hi,

Sorry for late response. This is the binary DSDT dump, teel me if it's the format that you want. 
[DSDT8_aml.zip](https://github.com/zeule/asus-ec-sensors/files/15225991/DSDT8_aml.zip)

I've asked about LibreHardwareMonitorLib because in the github repository of LibreHardwareMonitor, it is written in "LibreHardwareMonitorLib/Hardware/Motherboard/Lpc/EC/EmbeddedController.cs" 👍 
```
    // If you are updating board information, please consider sharing your changes with the corresponding Linux driver.
    // You can do that at https://github.com/zeule/asus-ec-sensors or contribute directly to Linux HWMON.
    // If you are adding a new board, please share DSDT table for the board at https://github.com/zeule/asus-ec-sensors.
    // https://dortania.github.io/Getting-Started-With-ACPI/Manual/dump.html
```

So I believed that you are also maintain the code for the EmbeddedController part of LibreHardwareMonitor. Am I wrong ?

zeule, 2024-05-17:
Can you build the driver and test changes? I haven't learned much from the DSDT, so I propose to start with copying over the sensor data for strix_z390_f_gaming. Do you prefer to test in Linux with this driver or on Windows with LHM?

FabXxXxXx, 2024-05-20:
I prefer to test on windows, but I don't know how to build the drivers. Is there a documentation somewhere to help me ?