zeule/asus-ec-sensors #52: Support for ROG STRIX B650E-F GAMING WIFI

psibi, 2024-05-26:
I tried to add support for it using the existing code, but wasn't successful. Relevant branch: https://github.com/psibi/asus-ec-sensors/tree/b650e

When I try to run sensors command with my branch, this is what I see:

``` shellsession
asusec-isa-0000
Adapter: ISA adapter
ERROR: Can't get value of subfeature temp1_input: I/O error
CPU:              N/A
ERROR: Can't get value of subfeature temp2_input: I/O error
CPU Package:      N/A
ERROR: Can't get value of subfeature temp3_input: I/O error
Motherboard:      N/A
ERROR: Can't get value of subfeature temp4_input: I/O error
VRM:              N/A
ERROR: Can't get value of subfeature temp5_input: I/O error
Water_In:         N/A
ERROR: Can't get value of subfeature temp6_input: I/O error
Water_Out:        N/A
```

This is my DSDT dump: 
[dsdt.tar.gz](https://github.com/zeule/asus-ec-sensors/files/15447077/dsdt.tar.gz)

Let me know if there is anything else I can add that will help.

zeule, 2024-05-26:
Hey!

Usually the ASUS boards provide sensors readings via EC when the number of sensors do not fit in the Super I/O registers. With this board I see only the T_Sensor, and it should fit. The DSDT contains no mention of the EC device either.

psibi, 2024-05-26:
Thanks!

Is there driver for T_Sensor that I can ? Right now all I see is this via `sensors`:

```
amdgpu-pci-0b00
Adapter: PCI adapter
vddgfx:        1.28 V  
vddnb:         1.01 V  
edge:         +40.0°C  
PPT:          42.23 W  

nvme-pci-0a00
Adapter: PCI adapter
Composite:    +33.9°C  (low  =  -5.2°C, high = +89.8°C)
                       (crit = +93.8°C)

k10temp-pci-00c3
Adapter: PCI adapter
Tctl:         +44.1°C  
Tccd1:        +38.5°C  
Tccd2:        +35.0°C  
```

zeule, 2024-05-26:
You need to find out the model of the Super I/O chip this board is using. The sensors-detect utility might help.

psibi, 2024-05-26:
Cool, thanks. Closing this issue then.

closed 2024-05-26 by psibi

zeule, 2024-05-28:
If you have and access to Windows, HWINFO shows chip names.

psibi, 2024-06-01:
I saw that the upstream linux supports this motherboard using NCT6775 driver: https://github.com/torvalds/linux/blob/cc8ed4d0a8486c7472cd72ec3c19957e509dc68c/drivers/hwmon/nct6775-platform.c#L1385

I was able to load it via modproble:

```
❯ sudo modprobe -v nct6775
[sudo: authenticate] Password:
insmod /run/booted-system/kernel-modules/lib/modules/6.6.32/kernel/drivers/hwmon/hwmon-vid.ko.xz
insmod /run/booted-system/kernel-modules/lib/modules/6.6.32/kernel/drivers/hwmon/nct6775-core.ko.xz
insmod /run/booted-system/kernel-modules/lib/modules/6.6.32/kernel/drivers/hwmon/nct6775.ko.xz
```

And then see `sensors` CLI report additional information like fan speed, various temperatures etc.

der-eismann, 2025-07-03:
I wish I would have found your issue earlier! I just got a new ASUS mainboard and was looking forward to contribute here, but after an hour of investigation it looks like there is simply no EC sensor on my TUF Gaming B650E-E WIFI either. I wonder if this is a general trend for newer models or just most of the non-ROG models that are left out.