zeule/asus-ec-sensors #57: CPU Amperage and temperature appear under same name for sensors.

RobotRoss, 2024-07-09:
Calling 'sensors' will return the list of supported hardware sensors, and is used by some applications to monitor the status of hardware devices.

Any item wanting to use the CPU temp however may get the amperage returned instead, as both appear under "asusec-isa-0000" : "CPU".

```
asusec-isa-0000
Adapter: ISA adapter
CPU Core:      1.24 V  
Chipset:     2726 RPM
Chipset:      +63.0°C  
CPU:          +52.0°C  
Motherboard:  +35.0°C  
T_Sensor:     -40.0°C  
VRM:          +20.0°C  
CPU:          21.00 A
```

It might be better to rename any identical listings (e.g. CPU, Chipset) to add an underscore and then their respective return value, e.g. "CPU_TEMP" , "CPU_POWER", "CHIPSET_TEMP", "CHIPSET_FAN", etc.

zeule, 2024-07-11:
Sorry, could you elaborate, please? I can't see how the sensors can be misread, because their value units differ. For the record, I tried to follow the names from the ASUS monitor in the UEFI firmware.

RobotRoss, 2024-07-11:
> Sorry, could you elaborate, please? I can't see how the sensors can be misread, because their value units differ. For the record, I tried to follow the names from the ASUS monitor in the UEFI firmware.

I've got 2 applications which I've used so far which obtain data from sensors using the chip name (in my case, "asusec-isa-0000") and the sensor name ("CPU"). The problem is, that because the CPU current and CPU temperature reading are both under the name "CPU" (and both programs will accept both units), I'll randomly see the amperage where I was expecting the temperature and vise-versa. (See my post above for what the output of `sensors` is)

Can be seen in Astra Monitor ( https://extensions.gnome.org/extension/6682/astra-monitor/ ) and OpenLinkHub ( https://github.com/jurkovic-nikola/OpenLinkHub )

KeithMyers, 2024-07-11:
Seems to me the problem is with your monitoring programs.  I have no issues with Psensor or GKrellm monitors.
They know the difference between amps, temps and volts for the cpu descriptors.