zeule/asus-ec-sensors #63: Modprobe not finding asus-ec-sensors after running dkms

reisholmes, 2025-05-15:
Hello, I'm trying to add in support for a new motherboard and experiencing an error at the modprobe step.

make:

```
❯ sudo LLVM=true make
  CC [M]  asus-ec-sensors.o
  MODPOST Module.symvers
  CC [M]  asus-ec-sensors.mod.o
  CC [M]  .module-common.o
  LD [M]  asus-ec-sensors.ko
  BTF [M] asus-ec-sensors.ko
```
modules_install:

```
❯ sudo LLVM=true make modules
❯ sudo LLVM=true make modules_install
  INSTALL /lib/modules/6.14.6-2-cachyos/extra/drivers/hwmon/asus-ec-sensors.ko
  SIGN    /lib/modules/6.14.6-2-cachyos/extra/drivers/hwmon/asus-ec-sensors.ko
At main.c:171:
- SSL error:FFFFFFFF80000002:system library::No such file or directory: crypto/bio/bss_file.c:67
- SSL error:10000080:BIO routines::no such file: crypto/bio/bss_file.c:75
sign-file: /usr/lib/modules/6.14.6-2-cachyos/build/certs/signing_key.pem
  ZSTD    /lib/modules/6.14.6-2-cachyos/extra/drivers/hwmon/asus-ec-sensors.ko.zst
  DEPMOD  /lib/modules/6.14.6-2-cachyos
```

dkms

```
❯ sudo LLVM=true make dkms_configure
❯ sudo LLVM=true make dkms
Creating symlink /var/lib/dkms/asus-ec-sensors/v0.1.0-67-g021d987/source -> /usr/src/asus-ec-sensors-v0.1.0-67-g021d987
Sign command: /usr/lib/modules/6.14.6-2-cachyos/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module(s)... done.
Signing module /var/lib/dkms/asus-ec-sensors/v0.1.0-67-g021d987/build/asus-ec-sensors.ko
Found pre-existing /usr/lib/modules/6.14.6-2-cachyos/kernel/drivers/hwmon/asus-ec-sensors.ko.zst, archiving for uninstallation
Installing /usr/lib/modules/6.14.6-2-cachyos/updates/dkms/asus-ec-sensors.ko.zst
Running depmod... done.
modprobe: ERROR: could not insert 'asus_ec_sensors': No such device
make: *** [Makefile:59: dkms] Error 1
```

I'm fairly new to this stuff so please let me know if I've done something wrong but it appears that modprobe doesn't see the new device.
Any tips or hints would be much appreciated.

zeule, 2025-05-15:
Have you already added your M/B model to the `dmi_table` array?

zeule, 2025-05-15:
This is likely where it exits: https://github.com/zeule/asus-ec-sensors/blob/master/asus-ec-sensors.c#L1046

reisholmes, 2025-05-15:
> Have you already added your M/B model to the `dmi_table` array?

Yes, dmidecode shows up as:

```
Base Board Information
	Manufacturer: ASUSTeK COMPUTER INC.
	Product Name: ROG MAXIMUS X HERO`
```

I added the following to the dmi_table array:

```
DMI_EXACT_MATCH_ASUS_BOARD_NAME("ROG MAXIMUS X HERO",
				&board_info_maximus_x_hero),
```

With my ec_board_info as the following (I just wanted to get T_Sensor temp for water, I don't care about the rest):

```
static const struct ec_board_info board_info_maximus_x_hero = {
	.sensors = SENSOR_TEMP_T_SENSOR,
	.mutex_path = ASUS_HW_ACCESS_MUTEX_ASMX,
	.family = family_intel_300_series,
};
```

zeule, 2025-05-15:
In that case I can only suggest injecting tracing printk() calls to understand why it exits with ENODEV

closed 2025-05-15 by reisholmes

zeule, 2025-05-15:
Glad you figured everything out. If you decide to share the incomplete board definition (just a single sensor), I would appreciate that.

reisholmes, 2025-05-15:
> Glad you figured everything out. If you decide to share the incomplete board definition (just a single sensor), I would appreciate that.

Sure. I’ll see what other sensors I can add from what’s available. HWInfo64 in Windows is only showing the following:

[image]

While they’re all working in HWInfo64, I can’t see exactly what the map to in ec-sensors. It looks like:

PCH = ec_sensor_temp_chipset

T_Sensor1 = ec_sensor_temp_t_sensor

CPU PLL = ???

VCCMP = this appears to be RAM voltage but I’m not seeing that in the list

CPU Core Current = ec_sensor_curr_cpu

CPU Core Power = ec_sensor_in_cpu_core , possibly? I will need to confirm later.

If you have any suggestions or corrections let me know. I’ll submit a PR later.