zeule/asus-ec-sensors #71: Support for TUF GAMING B850-PLUS WIFI mobo.

spinktvis, 2025-07-24:
Hi,

I have read through [https://github.com/zeule/asus-ec-sensors/blob/master/CONTRIBUTING.md](url) however I will probably need some assistance. The board is probably? similar to [https://github.com/zeule/asus-ec-sensors/pull/70](url) and [https://github.com/zeule/asus-ec-sensors/issues/69](url). Below is the relevant system info.

```
uname -a
Linux anarchy 6.15.7-3-cachyos #1 SMP PREEMPT_DYNAMIC Tue, 22 Jul 2025 15:54:52 +0000 x86_64 GNU/Linux
```

```
# dmidecode 3.6
Getting SMBIOS data from sysfs.
SMBIOS 3.6.0 present.
Table at 0x7769F000.

Handle 0x0000, DMI type 0, 26 bytes
BIOS Information
        Vendor: American Megatrends Inc.
        Version: 1028
        Release Date: 04/29/2025
        Address: 0xF0000
        Runtime Size: 64 kB
        ROM Size: 32 MB
        Characteristics:
                PCI is supported
                BIOS is upgradeable
                BIOS shadowing is allowed
                Boot from CD is supported
                Selectable boot is supported
                BIOS ROM is socketed
                EDD is supported
                Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
                Japanese floppy for Toshiba 1.2 MB is supported (int 13h)
                5.25"/360 kB floppy services are supported (int 13h)
                5.25"/1.2 MB floppy services are supported (int 13h)
                3.5"/720 kB floppy services are supported (int 13h)
                3.5"/2.88 MB floppy services are supported (int 13h)
                Print screen service is supported (int 5h)
                Serial services are supported (int 14h)
                Printer services are supported (int 17h)
                CGA/mono video services are supported (int 10h)
                ACPI is supported
                USB legacy is supported
                BIOS boot specification is supported
                Targeted content distribution is supported
                UEFI is supported
        BIOS Revision: 10.28

Handle 0x0001, DMI type 1, 27 bytes
System Information
        Manufacturer: ASUS
        Product Name: System Product Name
        Version: System Version
        Serial Number: System Serial Number
        UUID: e222f01a-cfd9-71ba-4b00-bcfce7196e01
        Wake-up Type: Power Switch
        SKU Number: SKU
        Family: To be filled by O.E.M.

Handle 0x0002, DMI type 2, 15 bytes
Base Board Information
        Manufacturer: ASUSTeK COMPUTER INC.
        Product Name: TUF GAMING B850-PLUS WIFI
        Version: Rev 1.xx
        Serial Number: 241045828401020
        Asset Tag: Default string
        Features:
                Board is a hosting board
                Board is replaceable
        Location In Chassis: Default string
[... 824 lines cut ...]
        Installable Languages: 9
                en|US|iso8859-1
                fr|FR|iso8859-1
                zh|TW|unicode
                zh|CN|unicode
                ja|JP|unicode
                de|DE|iso8859-1
                es|ES|iso8859-1
                ru|RU|iso8859-5
                ko|KR|unicode
        Currently Installed Language: en|US|iso8859-1

Handle 0x004E, DMI type 127, 4 bytes
End Of Table
```

```
grep -e ''  -n /sys/class/dmi/id/board_{name,vendor}
/sys/class/dmi/id/board_name:1:TUF GAMING B850-PLUS WIFI
/sys/class/dmi/id/board_vendor:1:ASUSTeK COMPUTER INC.
```

Attached are the original DSDT and decompiled. And now I am not sure on how to proceed. Can someone more tech savvy please assist in this matter? Or will this info and data together with the two other 800 series mobos threads be enough?

[dsdt.tar.gz](https://github.com/user-attachments/files/21412646/dsdt.tar.gz)

spinktvis, 2025-07-24:
Just loaded the ec_sys module. 

```
lsmod | grep "ec_sys"
ec_sys                 12288  0
```

Rerun the DSDT export and decompiled it (Not sure if a reboot is needed first.) 

```
cat dsdt.dsl | grep "Controller"
                    Name (_HID, EisaId ("PNP0000") /* 8259-compatible Programmable Interrupt Controller */)  // _HID: Hardware ID
                    Name (_HID, EisaId ("PNP0200") /* PC-class DMA Controller */)  // _HID: Hardware ID
                I2cSerialBusV2 (0x0008, ControllerInitiated, 0x00061A80,
                I2cSerialBusV2 (0x0042, ControllerInitiated, 0x00061A80,
                    I2cSerialBusV2 (0x004E, ControllerInitiated, 0x00061A80,

cat dsdt.dsl | grep "Embedded"

~

```
Not sure if that is the info you needed? And if that is so, and this mobo is out of scope for asus-ec-sensors, what would be the right place? Because Linux is unable to read any sensor data on this mobo outside of a few basics such as the CPU (provided by k10temp), GPU (provided by the Nvidia driver), and NVME, the rest is unavailable such as fan speeds, power delivery, VRM, HDDs and RAM.

spinktvis, 2025-07-24:
No file exists on that specific location. Any other places to check?

zeule, 2025-07-24:
Excuse me, to check for what? I believe the board does not have the EC.

spinktvis, 2025-07-24:
I didn't know that was the only location where it could exist...

Anyway, that is unfortunate it doesn't have the EC, apologies for creating this issue in the first place. Do you know if there is any other way to read all the sensor data? Are there any other projects similar for other Asus mobos? Or is this something that needs to be implemented into the kernel itself? I am not really familiar yet with the Asus ecosystem on Linux (only had this mobo for a couple of days now) and still trying to find my way around.

KeithMyers, 2025-07-24:
You can usually use the nct6775 driver to read board sensors in Linux.

spinktvis, 2025-07-24:
Thanks for the helpful feedback and hints! I added the nct6775 module, and now there are sensors being read, not sure if it is all of them, but at least it is something! Still at a loss as to where my HDD sensors went tho...

journal:
```
sudo dmesg | grep nct6775
[    2.194693] nct6775: Found NCT6796D-S/NCT6799D-R or compatible chip at 0x2e:0x290
```

sensors:
```
nct6799-isa-0290
Adapter: ISA adapter
in0:                          824.00 mV (min =  +0.00 V, max =  +1.74 V)
in1:                          992.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in2:                            3.41 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in3:                            3.26 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in4:                          992.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in5:                          872.00 mV (min =  +0.00 V, max =  +0.00 V)
in6:                          1000.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in7:                            3.41 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in8:                            3.26 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in9:                            3.26 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in10:                           1.44 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in11:                           1.12 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in12:                           1.03 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in13:                         1000.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in14:                           1.02 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in15:                         904.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in16:                           1.82 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in17:                           1.27 V  (min =  +0.00 V, max =  +0.00 V)
fan1:                            0 RPM  (min =    0 RPM)
fan2:                          948 RPM  (min =    0 RPM)
fan3:                         1192 RPM  (min =    0 RPM)
fan4:                          784 RPM  (min =    0 RPM)
fan5:                          934 RPM  (min =    0 RPM)
fan6:                          786 RPM  (min =    0 RPM)
fan7:                            0 RPM  (min =    0 RPM)
SYSTIN:                        +31.0°C  (high = +95.0°C, hyst = +48.0°C)  sensor = thermistor
CPUTIN:                        +33.0°C  (high = +80.0°C, hyst = +75.0°C)
                                        (crit = +125.0°C)  sensor = thermistor
AUXTIN0:                       +26.0°C  (high = +80.0°C, hyst = +75.0°C)
                                        (crit = +125.0°C)  sensor = thermistor
AUXTIN1:                        +3.0°C  (high = +80.0°C, hyst = +75.0°C)
                                        (crit = +125.0°C)  sensor = thermistor
AUXTIN2:                       +19.0°C  (high = +80.0°C, hyst = +75.0°C)
                                        (crit = +127.0°C)  sensor = thermistor
AUXTIN3:                       +25.0°C  (high = +80.0°C, hyst = +75.0°C)
                                        (crit = +125.0°C)  sensor = thermistor
SMBUSMASTER 0:                 +43.0°C  (high = +95.0°C, hyst = +48.0°C)
AUXTIN4:                       +33.0°C  (high = +80.0°C, hyst = +75.0°C)
                                        (crit = +100.0°C)
PECI Agent 0:                  +43.0°C  (high = +80.0°C, hyst = +75.0°C)
PECI/TSI Agent 0 Calibration:  +33.0°C  
AUXTIN5:                       +12.0°C  
PCH_CHIP_CPU_MAX_TEMP:          +0.0°C  
TSI0_TEMP:                     +43.9°C  
pwm1:                              87%  (mode = pwm)
pwm2:                              47%  (mode = pwm)
pwm3:                              87%  (mode = pwm)
pwm4:                              47%  (mode = pwm)
pwm6:                              47%  (mode = pwm)
pwm7:                             128%  (mode = pwm)
intrusion0:                   ALARM
intrusion1:                   OK
beep_enable:                  disabled
```

However, according to this comment in the following thread https://bugzilla.kernel.org/show_bug.cgi?id=204807#c329 support is still being worked on. Not sure how relevant that still is, since that was posted back in 2023. Anyway, thanks for the valued feedback!

zeule, 2025-07-24:
> Still at a loss as to where my HDD sensors went tho...

For SATA devices try the drivetemp module, NVME should show up automatically. You might want to try the spd5118 module for RAM stick temps.

spinktvis, 2025-07-24:
Both modules work!

```
drivetemp-scsi-0-0
Adapter: SCSI adapter
temp1:        +37.0°C  (low  = +14.0°C, high = +55.0°C)
                       (crit low = +10.0°C, crit = +60.0°C)
                       (lowest = +24.0°C, highest = +41.0°C)

drivetemp-scsi-1-0
Adapter: SCSI adapter
temp1:        +40.0°C  (low  =  +0.0°C, high = +55.0°C)
                       (crit low = -40.0°C, crit = +70.0°C)
                       (lowest = +22.0°C, highest = +44.0°C)

spd5118-i2c-6-51
Adapter: SMBus PIIX4 adapter port 0 at 0b00
temp1:        +31.8°C  (low  =  +0.0°C, high = +55.0°C)
                       (crit low =  +0.0°C, crit = +85.0°C)

spd5118-i2c-6-53
Adapter: SMBus PIIX4 adapter port 0 at 0b00
temp1:        +31.2°C  (low  =  +0.0°C, high = +55.0°C)
                       (crit low =  +0.0°C, crit = +85.0°C)
```

I will read up on the lm sensors documentation and see how far I am able to get. Since there's no Windows install to compare to it might be a little tricky, but we'll see. 

Again, thanks a lot for all the detailed info! Hopefully it will also be of some use for somebody else in the future with the same mobo and finds all this here. Closing this ticket.