zeule/asus-ec-sensors #76: ASUS PRIME Z270-A: anyone interested?

PerAstraAdDeum, 2025-08-19:
Hi there,

I've got access to an older Skylake/Kabylake mainboard, namely the [ASUS PRIME Z270-A](https://www.asus.com/us/motherboards-components/motherboards/prime/prime-z270-a/).

Before it is sold off I'd like to offer providing anything necessary for porting this board to the appropriate database for HWMON support in Linux (I hope I've come to the right place for this?) as an effort to increase the number of older ASUS boards being supported.

I'd do it myself, but I'm no dev and the contributing FAQ is fairly hard to understand for me. So, if anyone would be interested in guiding me through the process, or telling me how to gather the necessary data to do it themselves, I'd happily comply.

Best regards!

PerAstraAdDeum, 2025-08-30:
> We can start with DSDT dump, please.

Hey! I got the requested file. *.dsl files doesn't seem to be supported by Github though, so I've changed the file type to txt. Hopefully that doesn't change the content or how it's readable?

 
[dsdt.txt](https://github.com/user-attachments/files/22057428/dsdt.txt)

Also these, maybe they are helpful:

[dmidecode.txt](https://github.com/user-attachments/files/22057442/dmidecode.txt)
[iasl_dsdt.txt](https://github.com/user-attachments/files/22057443/iasl_dsdt.txt)

Best regards!

PerAstraAdDeum, 2025-09-01:
> Thanks! The board contains an EC and we can try to poke it. I have no idea what to expect, thus you would likely need to experiment and invest some time. I'd start with something similar to:
> 
> static const struct ec_board_info board_info_prime_z270_a = {
> 	.sensors = SENSOR_SET_TEMP_CHIPSET_CPU_MB |
> 		SENSOR_TEMP_T_SENSOR |
> 		SENSOR_TEMP_VRM |	SENSOR_FAN_CPU_OPT,
> 	.mutex_path = ACPI_GLOBAL_LOCK_PSEUDO_PATH, // more narrow lock can be figured out later
> 	.family = family_intel_300_series, // if the driver works, copy it into family_intel_200_series
> };

Hey! Thanks, I'll get to it immediately!

However, I'm not quite sure on what to do with this. Am I to save that code in a file and execute it?

Best regards!

PerAstraAdDeum, 2025-09-01:
> Ah, OK. Do you want a full diff, or a full changed asus-ec-sensors.c file?

I'm not sure. I assume the latter would be easier?

> You would need to clone this repository locally, compile the module (`$ make`), likely sign it (I use `$ sudo /usr/src/linux/scripts/sign-file sha512 /usr/src/linux/certs/signing_key.pem /usr/src/linux/certs/signing_key.x509 ./asus-ec-sensors.ko` here), then load the module (`$ sudo insmod ./asus-ec-sensors.ko`). Then you check `dmesg` for module loading errors, and if it loads correctly, you get sensor readings via `$ sesnsors 'asus-ec*'`.

That can be done by running `git clone https://github.com/zeule/asus-ec-sensors.git`, correct?

> To apply changes to the source code, you repeat all of that, but before inserting the module you unload it with `$ sudo rmmod asus-ec-sensors`.

Alright, then we should see if it picks anything up. Are you on Matrix or Discord by chance? I'd reckon this would probably be easier if I could reach you there instead of cluttering this issue here?

zeule, 2025-09-01:
> I'm not sure. I assume the latter would be easier?

If you don't want to use diff/patch, then yes, exchanging full files is easier.

> That first step can be done by running `git clone https://github.com/zeule/asus-ec-sensors.git`, correct?

Correct.

> Alright, then we should see if it picks anything up.

Here is the first draft, please:
[asus-ec-sensors.c](https://github.com/user-attachments/files/22078139/asus-ec-sensors.c)

>  Are you on Matrix or Discord by chance? I'd reckon this would probably be easier if I could reach you there instead of cluttering this issue here?

I'm at the office and can't answer quickly, sorry.

PerAstraAdDeum, 2025-09-01:
> Here is the first draft, please: [asus-ec-sensors.c](https://github.com/user-attachments/files/22078139/asus-ec-sensors.c)

Thank you! :)

Here's what I did now:

1. `git clone https://github.com/zeule/asus-ec-sensors.git`
2. Copied the file you've linked in your comment inside the `asus-ec-sensors` folder.
3. Ran `make`. Output:
```
 CC [M]  asus-ec-sensors.o
  MODPOST Module.symvers
  CC [M]  asus-ec-sensors.mod.o
  CC [M]  .module-common.o
  LD [M]  asus-ec-sensors.ko
  BTF [M] asus-ec-sensors.ko
Skipping BTF generation for asus-ec-sensors.ko due to unavailability of vmlinux
```
4. Tried to sign it, but that returned `command not found`. The folder `/usr/src/linux` doesn't exist on my system (Bazzite, based on Fedora Atomic).
5. Tried loading the module with `$ sudo insmod ./asus-ec-sensors.ko`. Output:
```
insmod: ERROR: could not insert module ./asus-ec-sensors.ko: No such device
```
6. Did some searching and tried `modinfo asus-ec-sensors` to see if the module was loaded. Output:
```
filename:       /lib/modules/6.16.4-103.bazzite.fc42.x86_64/kernel/drivers/hwmon/asus-ec-sensors.ko.xz
license:        GPL
description:    HWMON driver for sensors accessible via ACPI EC in ASUS motherboards
author:         Eugene Shalygin <eugene.shalygin@gmail.com>
alias:          dmi*:rvn*ASUSTeKCOMPUTERINC.*:rn*TUFGAMINGX670E-PLUS*:
alias:          dmi*:rvn*ASUSTeKCOMPUTERINC.*:rn*ROGZENITHIIEXTREMEALPHA*:
alias:          dmi*:rvn*ASUSTeKCOMPUTERINC.*:rn*ROGZENITHIIEXTREME*:
alias:          dmi*:rvn*ASUSTeKCOMPUTERINC.*:rn*ROGSTRIXZ690-AGAMINGWIFID4*:
alias:          dmi*:rvn*ASUSTeKCOMPUTERINC.*:rn*ROGSTRIXZ390-FGAMING*:
alias:          dmi*:rvn*ASUSTeKCOMPUTERINC.*:rn*ROGSTRIXX570-IGAMING*:
alias:          dmi*:rvn*ASUSTeKCOMPUTERINC.*:rn*ROGSTRIXX570-FGAMING*:
alias:          dmi*:rvn*ASUSTeKCOMPUTERINC.*:rn*ROGSTRIXX570-EGAMINGWIFIII*:
alias:          dmi*:rvn*ASUSTeKCOMPUTERINC.*:rn*ROGSTRIXX570-EGAMING*:
alias:          dmi*:rvn*ASUSTeKCOMPUTERINC.*:rn*ROGSTRIXB550-IGAMING*:
alias:          dmi*:rvn*ASUSTeKCOMPUTERINC.*:rn*ROGSTRIXB550-EGAMING*:
alias:          dmi*:rvn*ASUSTeKCOMPUTERINC.*:rn*ROGCROSSHAIRVIIIIMPACT*:
alias:          dmi*:rvn*ASUSTeKCOMPUTERINC.*:rn*ROGMAXIMUSZ690FORMULA*:
alias:          dmi*:rvn*ASUSTeKCOMPUTERINC.*:rn*ROGMAXIMUSXIHERO(WI-FI)*:
alias:          dmi*:rvn*ASUSTeKCOMPUTERINC.*:rn*ROGMAXIMUSXIHERO*:
alias:          dmi*:rvn*ASUSTeKCOMPUTERINC.*:rn*ROGCROSSHAIRX670EGENE*:
alias:          dmi*:rvn*ASUSTeKCOMPUTERINC.*:rn*ROGCROSSHAIRX670EHERO*:
alias:          dmi*:rvn*ASUSTeKCOMPUTERINC.*:rn*ROGCROSSHAIRVIIIHERO(WI-FI)*:
alias:          dmi*:rvn*ASUSTeKCOMPUTERINC.*:rn*ROGCROSSHAIRVIIIHERO*:
alias:          dmi*:rvn*ASUSTeKCOMPUTERINC.*:rn*ROGCROSSHAIRVIIIFORMULA*:
alias:          dmi*:rvn*ASUSTeKCOMPUTERINC.*:rn*ROGCROSSHAIRVIIIDARKHERO*:
alias:          dmi*:rvn*ASUSTeKCOMPUTERINC.*:rn*ProWSX570-ACE*:
alias:          dmi*:rvn*ASUSTeKCOMPUTERINC.*:rn*ProArtB550-CREATOR*:
alias:          dmi*:rvn*ASUSTeKCOMPUTERINC.*:rn*ProArtX670E-CREATORWIFI*:
alias:          dmi*:rvn*ASUSTeKCOMPUTERINC.*:rn*ProArtX570-CREATORWIFI*:
alias:          dmi*:rvn*ASUSTeKCOMPUTERINC.*:rn*PRIMEX670E-PROWIFI*:
alias:          dmi*:rvn*ASUSTeKCOMPUTERINC.*:rn*PRIMEX570-PRO*:
alias:          dmi*:rvn*ASUSTeKCOMPUTERINC.*:rn*PRIMEX470-PRO*:
alias:          dmi*:rvn*ASUSTeKCOMPUTERINC.*:rn*MAXIMUSVIHERO*:
depends:        
intree:         Y
name:           asus_ec_sensors
retpoline:      Y
vermagic:       6.16.4-103.bazzite.fc42.x86_64 SMP preempt mod_unload 
sig_id:         PKCS#7
signer:         Fedora kernel signing key
sig_key:        58:3D:C9:9D:31:6F:AF:3F:40:A3:ED:60:13:2B:BF:6F:9A:13:9E:5C
sig_hashalgo:   sha256
signature:      54:B8:B5:82:71:90:C6:2B:4C:90:C9:9E:C2:FF:0E:68:5C:5E:3F:89:
		25:59:E1:7D:B9:6C:3C:9F:3B:3D:32:A6:65:61:00:D7:E2:40:F8:03:
		31:37:7A:4F:E5:B1:BD:34:C5:25:3A:6B:E2:D1:AD:B4:8B:10:87:2C:
		7E:99:46:66:B2:D1:B4:EA:64:19:92:F1:65:EE:A5:F2:CA:FD:EF:8C:
		E0:B2:36:16:2B:AF:4A:5A:FE:CC:31:44:AE:19:3E:C8:67:07:5F:18:
		8E:D9:23:B4:20:D4:7D:8E:9D:A9:5F:E4:91:9C:32:D6:54:0B:EB:0D:
		4B:D8:37:84:8F:59:CC:85:61:49:67:74:F5:6F:11:DA:45:7D:C4:D7:
		4E:9A:1F:D1:4B:7C:07:6A:DD:C8:CA:B1:60:6B:F2:DC:AA:5E:E7:CE:
		E3:34:47:AC:90:65:E4:8D:F5:B7:D7:E7:AA:23:15:94:B1:9F:7B:FC:
		56:97:60:E5:E8:06:98:65:87:CA:73:1D:94:D8:7C:EE:5C:11:9C:FF:
		BB:85:CE:72:BF:DE:0A:9B:C1:45:14:5C:32:F0:F1:A8:A0:81:2D:C2:
		A7:8F:02:72:2A:24:69:E0:8F:3C:03:31:2E:30:88:A7:ED:5A:D7:C2:
		B3:34:C1:E2:B3:BB:B4:0F:CF:F4:4E:EA:C1:13:3F:74:18:6B:1A:9E:
		D3:BF:28:DA:41:BB:24:0F:34:2C:B3:AA:FE:0D:11:0F:72:7A:CE:BF:
		67:46:5E:AD:82:0F:DD:81:7F:C1:2E:A3:D1:4C:79:BB:51:FC:CC:F1:
		7C:BB:B7:74:2D:D9:84:84:4F:F4:22:67:A7:8E:4E:D0:6E:30:13:23:
		D8:13:14:73:5B:1C:4F:90:CA:71:E1:94:6A:07:D1:63:C4:24:0A:1E:
		06:48:43:42:28:6C:05:7C:26:8C:33:55:BC:BB:17:67:41:E6:DF:24:
		F2:1F:5B:8D:22:2E:71:C3:F8:35:33:E5:2D:6C:AD:90:77:54:D5:D8:
		89:08:A7:D2:5A:A1:0F:24:AC:CF:E2:68:5D:1C:7E:BB:68:14:05:96:
		40:E2:CE:3B:3D:F8:18:D3:17:F5:A8:99:4E:61:34:62:97:13:3A:E2:
		AF:01:9B:7F:59:CD:89:F0:66:95:A9:E8:01:D0:A5:FD:AC:DD:3B:00:
		7A:CE:83:4A:7B:B3:35:04:E2:FD:C6:DA:1D:71:61:5C:0F:9C:FA:C1:
		C0:1E:4F:3C:93:91:D1:7A:EC:40:99:97:C5:DE:6C:4E:88:99:88:29:
		32:28:FA:E5:89:C7:31:BB:63:0A:85:73:C1:AF:7D:DD:16:17:AA:BB:
		2A:A1:85:E9:39:E4:90:10:53:B9:A4:DC
parm:           mutex_path:Override ACPI mutex path used to guard access to hardware (charp)
```
7. But apparently, it isn't loaded. `sudo rmmod asus-ec-sensors` returns `rmmod: ERROR: Module asus_ec_sensors is not currently loaded`.

How should I proceed?

> > Are you on Matrix or Discord by chance? I'd reckon this would probably be easier if I could reach you there instead of cluttering this issue here?
> 
> I'm at the office and can't answer quickly, sorry.

Sure, no problem! I'll report back here, then.

PerAstraAdDeum, 2025-09-01:
> Nice! Regarding the CPU_Opt it is either not served via EC registers (and its RPMs are reported elsewhere), or we try to read it from wrong register. If you have a rough estimate of the RPMs, or can change it, it is possible to find the register by dumping all of them via the ec_sys module. Load the module, and then use the hexdump utility to read registers (from the first bank only, but CPU_Opt should be in the first bank). Fans are reported as two byte values in RPMs, but since we don't know the offset, we can't just use `hexdump -d`. Instead, you can take calculator, convert approximate RPMs into hex and look for the higher byte. The lower byte for fans should fluctuate a bit (I mean between hexdump executions), unlike the high one, which should be stable.
> 
> Example: I guess fan RPMs are between 900 and 1200, which are 0x0384 and 0x04b0 in hex. I then look for 0x03 and 0x04 in hexdump ouput.

Delighted, however I'm afraid I'm not clear on the instructions. :D 

For starters, I've tried `sudo modprobe ec_sys` but that returned nothing?

The fan I'm using for testing is a Noctua NF-A9 PWM chromax.black.swap, 92mm with a alleged RPM range from 400-2000. I'd assume that since the fan speed is set to `standard` in BIOS your estimate of a RPM range of 900 to 1200 seems accurate?

zeule, 2025-09-01:
> For starters, I've tried `sudo modprobe ec_sys` but that returned nothing?

No errors can mean it was loaded successfully. The module creates "/sys/kernel/debug/ec/ec0/io" file you can read EC register content from, i.e. `# hexdump -c /sys/kernel/debug/ec/ec0/io`.

> The fan I'm using for testing is a Noctua NF-A9 PWM chromax.black.swap, 92mm with a alleged RPM range from 400-2000. I'd assume that since the fan speed is set to `standard` in BIOS your estimate of a RPM range of 900 to 1200 seems accurate?

There are stroboscope effect based apps for both Android and iOS to measure fan RPMs.

PerAstraAdDeum, 2025-09-02:
> No errors can mean it was loaded successfully. The module creates "/sys/kernel/debug/ec/ec0/io" file you can read EC register content from, i.e. `# hexdump -c /sys/kernel/debug/ec/ec0/io`.
>
>
> There [...]

Hey, I'm back at it and I've done some readings, however they are inconclusive. First, the output of `hexdump -c /sys/kernel/debug/ec/ec0/io` with **no fan attached**:

```
0000000  \0  \0  \0 001  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0
0000010  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0
*
0000030  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0   $   " 033 330  \0  \0
0000040   |  \0  \0  \r  \v 017 017 377  \0  \0  \0  \0  \0  \0  \0  \0
0000050  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0
0000060 200 200 200  \0  \0  \0  \0  \0  \a  \0  \0  \0  \0  \0  \0  \0
0000070  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0
0000080  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0   3   3 031 340
0000090 001   C 001   ` 001   ? 002   [ 001   M 003 375  \0  \0  \0  \0
00000a0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0
*
00000d0   @  \0 020 001 004  \0  \0  \0   3   3   D   C  \0  \0 004 001
00000e0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0
00000f0  \0 314  \b  \0  \a  \0  \0  \0  \0  \0  \0  \0  \0 335  \b  \0
0000100
```

Now, the same command but with the **Noctua fan attached**:

```
0000000  \0  \0  \0 001  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0
0000010  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0
*
0000030  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0   %   # 034 330  \0  \0
0000040   |  \0  \0  \r  \v 017 017 377  \0  \0  \0  \0  \0  \0  \0  \0
0000050  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0
0000060 200 200 200  \0  \0  \0  \0  \0  \a  \0  \0  \0  \0  \0  \0  \0
0000070  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0
0000080  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0   3   3 031 340
0000090 001   C 001   ` 001   ? 002   Q 001   M 003 375  \0  \0  \0  \0
00000a0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0
00000b0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0 002 356  \0  \0
00000c0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0
00000d0   @  \0 020 001 004  \0  \0  \0   3   3   D   C  \0  \0 004 001
00000e0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0  \0
00000f0  \0 314  \b  \0 004  \0  \0  \0  \0  \0  \0  \0  \0 335  \b  \0
0000100
```

At first I'd thought they're identical, but there are indeed a few differences here!

About the RPM monitoring: I hadn't had any luck with my measurements. Most apps I've tried with my Android phone were either designed with turntables in mind or didn't produce consistent readings (or didn't catch the fan's motion at all). However, the ASUS firmware reports the fan speed to be between 860 and 890, mostly hovering around 872 RPM.

Hope this helps!

zeule, 2025-09-03:
Thank you for the dumps! I'm sorry, I've made a mistake in the request: it should be `hexdumpc -C`, and not "-c".

Anyway, I guess the CPU_Opt fan is at 0xbc, and here is an updated source file. Could you test it, please?

[asus-ec-sensors.c](https://github.com/user-attachments/files/22114286/asus-ec-sensors.c)

PerAstraAdDeum, 2025-09-03:
> Thank you for the dumps! I'm sorry, I've made a mistake in the request: it should be `hexdumpc -C`, and not "-c".
>
> Anyway, I guess the CPU_Opt fan is at 0xbc, and here is an updated source file. [...]

Hi! No problem, tested the new file right away:

```
asusec-isa-0000
Adapter: ISA adapter
CPU_Opt:      730 RPM
Chipset:      +40.0°C  
CPU:          +33.0°C  
Motherboard:  +29.0°C  
T_Sensor:     -40.0°C  
VRM:           +0.0°C
```

Success! However, I'm under the impression that the fan speed reported by the driver is off by around ~100 RPM. Check these two screenshots for comparison, one is from the firmware and the other from the driver (Mission Control App):

[image]

[image]

Also, **CPU fan** speed isn't being reported.

I've done testing for CHA_FAN1 already. This is the output of `sudo hexdump -C /sys/kernel/debug/ec/ec0/io` **with no fan attached**:

```
00000000  00 00 00 01 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000010  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000030  00 00 00 00 00 00 00 00  00 00 29 22 1e d8 00 00  |..........)"....|
00000040  7c 00 00 0d 0b 0f 0f ff  00 00 00 00 00 00 00 00  ||...............|
00000050  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000060  80 80 80 00 00 00 00 00  07 00 00 00 00 00 00 00  |................|
00000070  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000080  00 00 00 00 00 00 00 00  00 00 00 00 33 33 19 e0  |............33..|
00000090  01 43 01 60 01 3e 02 2f  01 4d 03 fd 00 00 00 00  |.C.`.>./.M......|
000000a0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
000000d0  40 00 10 01 04 00 00 00  33 33 44 43 00 00 04 01  |@.......33DC....|
000000e0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
000000f0  00 cc 08 00 01 00 00 00  00 00 00 00 00 dd 08 00  |................|
00000100
```

This is the same command with the **Noctua fan attached** to CHA_FAN1:

```
00000000  00 00 00 01 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000010  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000030  00 00 00 00 00 00 00 00  00 00 29 22 1e d8 00 00  |..........)"....|
00000040  7c 00 00 0d 0b 0f 0f ff  00 00 00 00 00 00 00 00  ||...............|
00000050  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000060  80 80 80 00 00 00 00 00  07 00 00 00 00 00 00 00  |................|
00000070  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000080  00 00 00 00 00 00 00 00  00 00 00 00 33 33 19 e0  |............33..|
00000090  01 44 01 60 01 3f 02 2f  01 4d 03 fd 00 00 00 00  |.D.`.?./.M......|
000000a0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
000000d0  40 00 10 01 04 00 00 00  33 33 44 43 00 00 04 01  |@.......33DC....|
000000e0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
000000f0  00 cc 08 00 0e 00 00 00  00 00 00 00 00 dd 08 00  |................|
00000100
```

Thank you for your efforts again. Hope this helps.

Best regards

zeule, 2025-09-03:
Yes, Nuvoton NCT6793D is the chip which provides the hardware monitoring (ACPI EC physically is this very chip), you need the nct6775 module to access the information. The module might refuse to load due to https://bugzilla.kernel.org/show_bug.cgi?id=204807.

There are potentially two sensors at 0x3a and 0x3b, we currently read the one at 0x3a as Motherboard and at  0x3b as CPU temperatures. If the firmware name it CPU Package, we should do the same to not confuse people.

zeule, 2025-09-03:
There seems to be neither CPU_Package nor VRM  temperature readings in the BIOS. I guess we remove the VRM one and we are done. Thank you for the contribution!

zeule, 2025-09-03:
I'd like to get "Tested-by:" tag from you to submit this to the kernel. If you agree, please send me your real name and email (you can get my email from the commits in this repository).

PerAstraAdDeum, 2025-09-03:
> There seems to be neither CPU_Package nor VRM temperature readings in the BIOS. I guess we remove the VRM one and we are done. Thank you for the contribution!

Sure thing! Wan't me to test it one last time before closing this issue? To be sure that the VRM reading is really gone?

I'd like to get "Tested-by:" tag from you to submit this to the kernel. If you agree, please send me your real name and email (you can get my email from the commits in this repository).

Sure, will write you an email then.

zeule, 2025-09-03:
Oops, I forgot one more task: I should try to find an ACPI mutex to use while querying the EC (the current code uses the global ACPI lock, which is not good). The testing to do is to repeatedly call `sesnors` and check the kernel log for [this](https://github.com/zeule/asus-ec-sensors/blob/master/asus-ec-sensors.c#L984) message. Would you, please, do that if I find the mutex?

zeule, 2025-09-03:
Here is the updated source, please: 

[asus-ec-sensors.c](https://github.com/user-attachments/files/22123527/asus-ec-sensors.c)

Please check the kernel log for errors upon inserting the new module. There might be error "Failed to get hardware access guard AML mutex..." if I get the mutex name wrong, and already mentioned "Concurrent access to the ACPI EC detected.\nRace condition possible." after running `sensors` if this is not the right mutex to use.