zeule/asus-ec-sensors #83: ROG STRIX X870-F GAMING WIFI

pierre-guillot, 2025-09-22:
Hi @zeule !

Just get my new Asus Board and want to have a good sensor for work with coolercontrol (especially with VRM temp).

Here is some informations : 

> grep -e ''  -n /sys/class/dmi/id/board_{name,vendor}
> /sys/class/dmi/id/board_name:1:ROG STRIX X870-F GAMING WIFI
> /sys/class/dmi/id/board_vendor:1:ASUSTeK COMPUTER INC

I read instruction about "Adding a new motherboard" and if I understand, you need DSDT information :

[dsdt.zip](https://github.com/user-attachments/files/22472111/dsdt.zip)

I confirm I can compile the actual sources by using "make"  but if I try `modprobe ./asus-ec-sensors.ko`, I have the error `could not insert 'asus_ec_sensors': No such device` because the module doesn't known my board and that why I need to add it in the `dmi_table` in `asus-ec-sensors.c` ?

Important note : My Archlinux is in BIOS and not UEFI.

Use lot of chassis FAN and one CPU FAN on it.

pierre-guillot, 2025-09-22:
Just seeing that a strix_x870e_e_gaming_wifi branch and compiled it. But I still have `modprobe: ERROR: could not insert 'asus_ec_sensors': No such device` (sorry, I'm not a kernel module specialist).

zeule, 2025-09-22:
But of course the source code needs to be modified to make it work with a new board model. I pushed feature/x870f-gaming-wifi branch, could you, please, try that out?

zeule, 2025-10-28:
Please excuse my long silence, I had no time for my projects in the past months. 

I've tried to locate that chipset sensor and failed.  My proposal we submit what we have and then I finish the update of the ec_sys driver, which allows to read multiple banks and use mutex, and we can try again.

pierre-guillot, 2026-04-01:
asus-ec-sensors is just arrived on Archlinux for my motherboard : 

[image]

Many Thanks again for the work 👍 .

Has said before, I can start again to run some commands for find the chipset temperature if you want.

Regards